prusa3d/Prusa-Firmware #3672: 3.12.0 BETA1 and Revo 6 Extruder - THERMAL ANOMALY

madproducts, 2022-10-08:
Hi - Upgraded my MK3S+ to 3.12.0 BETA1 and tried to calibrate using M310 A F1 as per instructions on GiHub.  The calibration routine fails within a second or two with a message "THERMAL ANOMALY". At this point I am just reporting this in the forum. My opinion is that the heating process is so fast with a Revo 6 that the calibration process is detecting it as an anomaly. From 29 C the temp jumps to almost 50-55C in just two to three seconds and the error halts the process.  Will wait for a fix. 

Note: I did perform a PID calibration from the menu option before trying it, just for good measure. 

Keep up the good work Prusa Team!!!

arekm, 2022-10-08:
There is already a issue which gathers all information about revo thermal characteristics:
https://github.com/prusa3d/Prusa-Firmware/issues/3636

Please read it, produce serial.log and attach to that issue. Close this one to avoid duplication. Thanks.

madproducts, 2022-10-08:
Apologies. Closing. Did not notice before i created. Attaching serial logs 3636

madproducts, 2022-10-08:
Adding serial logs to #3636

closed 2022-10-08 by madproducts

arekm, 2022-10-08:
Note this is the procedure for gathering serial log:
https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1262926997