prusa3d/Prusa-Firmware #3676: [BUG]<Filament unload distance isn't long enough with FS cover for Prusa Enclosure>

tjhawkey, 2022-10-09:
Prusa MK3S+, firmware 3.12 beta, printing from Obico
I've noticed that when the filament runs out it seems like the printer doesn't stop printing until the filament has moved down towards the gears. When you unload to change the spool, there's not enough filament left to get out of the longer FS cover. I don't know yet if this is more related to slow response from the filament sensor, or insuffucient un-load with the new FS cover (or a bit of both)

Prusa-Support, 2022-10-12:
Hello, thank you for your report! From the description, in some cases it might be normal, there are cases where we might need to open up and extract the finished filament manually, though a possible test might be testing it without Obico-Octoprint to see if that way it might be faster. The manual removal is also covered in the fil. runout guide itself: https://help.prusa3d.com/article/ir-filament-sensor-mk2-5s-mk3s_112214

But to make sure, how did you mean by FS-cover for Prusa Enclosure? I don't think a modified FS-cover would influence this aspect, but how was the part different from the standard MK3S+ FS-cover? 

Alessandro Pantaleo
Prusa Research

darraghbr, 2022-10-12:
Is this the FS cover in question?

https://www.printables.com/model/251931-original-prusa-enclosure-printable-parts/files

The part labelled "FS-cover.stl" in the "LCD support & FS cover" folder.

tjhawkey, 2022-10-12:
Yes, that's the one (with fitting)

Prusa-Support, 2022-10-12:
Saw the difference now, sorry for the earlier confusion from my side on this. It might have to do with the extra length given the part and also the fitting on top of it. I'll put this to our devs attention, perhaps something to slightly correct to cover the cases where the different cover+fitting are there! 

Alessandro Pantaleo
Prusa Research

darraghbr, 2023-08-10:
I think the general solution to this issue is something along the lines of this piece of hardware here:

[Filament Sensor Cover with Magnetic coupling on Printables by Minimayhem](https://www.printables.com/model/365830-filament-sensor-cover-with-magnetic-coupling)

Although I think that specific implementation could use a little work in the Design for Additive Manufacture department.

closed 2024-01-15 by github-actions[bot]