prusa3d/Prusa-Firmware #3679: [BUG]3.12.0-ccf0c0a MMU2:Communication timeout

kronflux, 2022-10-10:
**Printer type** - MK3S+
**Printer firmware version** -  3.12.0-ccf0c0a (Compiled from source)

**MMU upgrade** - NONE
**MMU upgrade firmware version** - NA

**SD card or USB/Octoprint**
USB/Octoprint

**Describe the bug**
Printer outputs: `Recv: echo:MMU2:>S0*c6.` every second for a while then finally Emergency Stops with this message:
`Recv: Error:MMU2:Communication timeout, last bytes: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"`

leptun, 2022-10-10:
The MK3 branch is used for development. If you want a stable build please use one of the branches for specific versions.

If you want a quick workaround, disable the mmu in Settings and reset the printer.

kronflux, 2022-10-10:
Thank you. I had tried that but it was still doing it. After an M500 and a reboot it seems resolved.
Is there a way to disable MMU by default in source? Thanks!

closed 2022-10-10 by kronflux

leptun, 2022-10-10:
I don't think it's possible at the moment, but it should be possible before the 3.13 release.
The error on startup when no mmu is detected should probably also be suppressed. Or just assume the mmu is disabled by default.