prusa3d/Prusa-Firmware #3700: [ENHANCEMENT]<Out of filament change>

Mbrinch, 2022-10-18:
**Enter what type of printer or upgrade the enhancement applies to.**
MK3S, MK3

**Is your enhancement related to a problem? Please describe.**
When the printer runs out of filament, it moves the position of the printhead all the way to one side of the printer.
When it unloads the filament and the last bit gets stuck in the extruder, you need to open the side lit to get the filament out manually.
There is not enough room for removing anything that is stuck in there when the printhead is positioned all the way to the side. 
You need to push the print head to the other side in order to have enough room to get a tool in and remove the stuck filament bit.
If you push the printhead to the side, to have more workspace, the printer does not know where to start from when the new filament is loaded and you end up having a failed print.
  
**Describe the solution you'd like**
It would be really nice that when the printer detect it run out of filament, that it positions it self in the middle of the x axis  instead of all the way to the side.

leptun, 2022-10-18:
If you answer "No" to the "Was filament unload successful?" message, the X axis moves 100mm to the left

Prusa-Support, 2022-10-29:
Please @Mbrinch clarify if you actually selected "no", as that may be all about it. If that solves the problem, please close this issue.

Michele Moramarco
Prusa Research

closed 2023-11-09 by github-actions[bot]