prusa3d/Prusa-Firmware #3705: Z axis goes beyond the top (Prusa Mini)

backslashV, 2022-10-19:
I was moving the Z axis using the menu on the device and noticed that it tried to go beyond the top limit of the Z rod. I immediately turned off the printer as the motor was struggling. I didn't try other axes, but the firmware should disallow this IMO to prevent damage to the printer.

Prusa-Support, 2022-10-19:
Hello, thank you for the report. I don't believe this is causing damage or is a defect, but to make sure, please send me a video of this, ideally via email to me info@prusa3d.com mentioning me in the subject, so that we can start a thread there. If I understood the situation correctly, this happens as the printer takes as (0,0,0) point the point it is when starting up. This is then automatically corrected either with auto-homing the printer, or just before starting any print. I await the video to confirm the info, thank you. 

Alessandro Pantaleo
Prusa Research

3d-gussner, 2022-10-20:
Moving to https://github.com/prusa3d/Prusa-Firmware-Buddy