prusa3d/Prusa-Firmware #3707: [BUG]Thermal Anomaly Error occurs without a reason in BETA 3.12 firmware

KokuxMilsch, 2022-10-19:
Printer type - [MK3S]
Printer firmware version -  [BETA 3.12]

Printed via Octoprint.

I installed the new beta firmware to combat a temperature fluctuation issue I had. It worked fine after I calibrated the Thermal Model, but only once. The next time I started a print, the printer kept reporting a THERMAL ANOMALY error and eventually triggered an emergency stop.
  
The printer was in an enclosure during the first print. I moved it outside because I was upgrading the enclosure. I thought the THERMAL ANOMALY error was reasonable, because all the heat was dissipated to the environment much easier now. However, when I moved the printer back into the enclosure, it still kept reporting the THERMAL ANOMALY error. I didn't change anything related to thermal behavior in the enclosure.
  
Since I calibrated the Thermal Model inside the enclosure, it should still work.
  
All of the mentioned prints were sliced to be printed at 250°C (Prusament PETG).

arekm, 2022-10-19:
Stock MK3S, without any mods?

Try to capture serial log via octoprint (using the same method as described in other bug report - https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1262926997 ) from calibration *and* from actual run then attach log, with timestamps here.

KokuxMilsch, 2022-10-20:
Unfortunately I can't provide the logs now, because after starting another print, smoke started rising from the extruder. The print was also burnt. I immediately disconnected the power and poured some water over it to stop the fire. I will order a new v6 hotend now and I also wont touch the printer until then because this was quite a shocking experience.

closed 2022-10-20 by KokuxMilsch