prusa3d/Prusa-Firmware #3709: [BUG] Bed starts heating when selecting unload filament.

GMeyerGMeyer, 2022-10-19:
--> MK3S, 3.11, no MMU, Sd card,

As described, when i want to unload Filament, i do not need to heat the bed. But in fact, the printer does it. 

Wanted behaviour: When unloading Filmanet (PLA), do not start heating the bed. Just heat the nozzle. Thanks. 

Cheers George

neophyl, 2022-10-20:
Duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/422

Prusa-Support, 2022-10-20:
Hello, thank you for the post, I confirm it's a duplicate of #422, this enhancement is being discussed with devs, nothing is promised but perhaps there might be a future change. Closing this particular thread being it duplicate, but feel free to add more comments on the #422 thread. Thank you. 

Alessandro Pantaleo
Prusa Research

closed 2022-10-20 by Prusa-Support