prusa3d/Prusa-Firmware #3715: [BUG]3.12.0-BETA1 Inconsistent long-press for Live-Z behavior

zbig-t, 2022-10-24:
I'm testing 3.12.0-BETA1 on my MK3S+. Please note I'm also using PrusaLink 0.7RC1 so this behavior could be related to that (due to the changes on how LCD bottom line messages are being handled that were introduced in 3.12).

Printer's behavior back on 3.11:
- After encoder button long-press, the Live-Z menu was consistently popping up
- After some delay, where no Live-Z changes were being dialed-in for some time, the Live-Z menu would disappear.

Current behavior on 3.12.0.BETA1, when I print via PrusaConnect and the LCD bottom line scrolls the name of the file being printed:
- First encoder button long-press gets ignored, only resulting in what looks like an LCD redraw
- Second long press brings up the Live-Z menu, but then:
- Usually it doesn't ever time out, or:
- Sometimes times-out prematurely.

Prusa-Support, 2022-10-26:
Hello! Thank you for the report, I passed this on to our devs for them to reproduce and find a resolution to this. We'll update here in case of more info from our side. 🙂 

Alessandro Pantaleo
Prusa Research

gudnimg, 2022-11-06:
Note this issue is not related to PrusaLink.

The issue is due to a timeout timer not being restarted when processing the long-press event. In most cases this causes the menu to stay on the LCD indefinately. In some situations it is possible the menu is dismissed as soon as one enters it. 

The issue was likely introduced here during 3.12 development (by me 😄) https://github.com/prusa3d/Prusa-Firmware/pull/3268/files . The reason for this change is that prior to FW 3.12 the timer is restarted in an interrupt context. This is incorrect usage because it would be restarted multiple times for a single button event and so wastes clock cycles when serving the interrupt. Now (in 3.12) the timer is restarted outside the interrupt (and only once) when we know the LCD knob was clicked. 

I have submitted a fix for this issue here https://github.com/prusa3d/Prusa-Firmware/pull/3737 thanks for reporting it @zbig-t ! 👍

commit e9b7763a (2022-11-07): Fix long-press LCD inactivity timeout

Fixes #3715

When a long-press is triggered, the LCD inactivity timer
needs to be restarted. This is done with lcd_timeoutToStatus.start()

This also fixes situations where a long-press is triggered, and the timer
times out immediately (because it was not reset properly)

Change in memory:
Flash: +8 bytes
SRAM: 0 bytes
PR #3737 (2022-11-08): Fix long-press LCD inactivity timeout
Fixes #3715

Issue was likely introduced during 3.12 development here: https://github.com/prusa3d/Prusa-Firmware/pull/3268

When a long-press is triggered, the LCD inactivity timer needs to be restarted. This is done with `lcd_timeoutToStatus.start()`. The function `menu_lcd_lcdupdate_func()` does not reset the timer in this scenario so we need to do it in `menu_lcd_longpress_func()`

This also fixes unpredictable case where a long-press is triggered, and the timer times out immediately (because it was not reset properly)

Change in memory:
Flash: +8 bytes [... cut]
commit 7fe22e8f (2022-11-08): Fix long-press LCD inactivity timeout

Fixes #3715

When a long-press is triggered, the LCD inactivity timer
needs to be restarted. This is done with lcd_timeoutToStatus.start()

This also fixes situations where a long-press is triggered, and the timer
times out immediately (because it was not reset properly)

Change in memory:
Flash: +8 bytes
SRAM: 0 bytes
closed 2022-11-08 by 3d-gussner