prusa3d/Prusa-Firmware #3719: [FEATURE REQUEST]  MK3s FW 3.12 Beta 1 Option to temporary Disable Temperature model, when changing nozzle

extremeall, 2022-10-27:
When changing nozzle with the temperature model enabled, you get a warning. This makes sense because the temperature fluctates. 

However a feature to temporary disable the check when changing nozzle would be great. 

Maybe even a nozzle changing menu item, that preheats the extruder to the correct temperature and disables the check

F1rst-Lay3r, 2022-12-05:
That is an option I would like too. When changing the nozzle, you will now receive a 'Thermal Anomaly' warning. If it takes longer to change the nozzle, then an M112 follows. In itself great that the temperature model is doing its job, in case of a nozzle swap perhaps less desirable.

Prusa-Support, 2022-12-08:
Thanks for your feedback.

For now, when needed, please consider sending codes `M310 S0` and `M500` to _disable thermal model protection_ and _save_, as per https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1315241854.

Michele Moramarco
Prusa Research

wavexx, 2022-12-12:
This should be fixed by the new "Nozzle change" menu #3774

Prusa-Support, 2023-01-04:
The new FW [3.12.0-RC2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.0-RC2) solves and closes this issue.
This is an almost definitive version so feel free to help with the final testing and share feedback here.
The issue could be re-opened if necessary.

Michele Moramarco
Prusa Research

closed 2023-01-04 by Prusa-Support

F1rst-Lay3r, 2023-01-05:
The 'Nozzle change'  menu option is a great solution to temporarily disable the temperature model.
One thing comes to mind: 
When selecting the option, the printer first homes and then the extruder moves to a high position for better access to the extruder. That's a nice feature also, but be aware that the online manual 'instructs' to temporarily store the print fan between the timing belt during a nozzle change. It could happen that one has already removed the print fan and stored it between the belts, prior to choosing the menu-option. This way, the belts will move while the print fan is clamped between the belts. Maybe that should be mentioned in the online manual. 

Above all: thank you for adding this feature!