prusa3d/Prusa-Firmware #3722: [QUESTION]<Epositions calculation in Marlin code>

fbuna, 2022-10-28:
hello, 
I want to know how do I get the individual E positions out when approaching the individual points in the X, Y and Z direction (GCode Marlin)? So how does the GCode get all the exact EXxx values? Where can I find this calculation for the different EPositions (how much filament in each new x-,y- ,,, positions will be needed) in Marlin Code?

Thx :)

Prusa-Support, 2022-11-03:
Hello. I'm not really sure I understand your question. I beg you pardon in advance if I am not getting the point.

The g-codes that make the E-axis motor extrude, and any other X/Y/Z-axes motor change axis position, are G0 or G1.
A single line code can move multiple motors, which means spatial movement combined with filament extrusion where needed.
The calculation of the spatial and extruding movements is up to Prusa Slicer.

You can read how these codes work in [this article](https://help.prusa3d.com/article/prusa-specific-g-codes_112173) of the Help section.

I hope this answers your question.
If so please close the issue. If not, please clarify.

Michele Moramarco
Prusa Research

fbuna, 2022-11-03:
First of all, thank you for your reply. I already know the site I don't mean the GCode commands/definitions, but the code behind it in the Slic3r for calculating the Evalues of each new positon or in the Malrin firmware to control the extruder stepper (exact code calculations in C++ I need). I can calculate the values manually based on filament length, nozzle diameter, etc., but I do not find them in the firmware Marlin, which would have to be there or in the Slic3r code, which calculates it.

Prusa-Support, 2022-11-09:
Prusa Slicer can calculate how many millimeters the X/Y/Z/E stepper motors should move, based on

- spatial coordinates = distance between two points of a segment [mm],
- plastic volume = extrusion width x layer thickness [mm³],
- speed = necessary time to cover the distance of a segment [mm/s],
- plastic extrusion flow = volume of plastic released over time [mm³/s].

Prusa Slicer will simply translate this information into millimeters and feedrate (if needed) per each motor. If the feedrate is not specified, the default value will be used. The default feedrate value is pre-set in the firmware for each stepper motor based on the stepper motors calibration: this way the firmware already knows how many times the stepper motor should spin to move an axis, or the filament, by a certain amount of millimeters.

The stepper motor calibration is not needed and not recommended for genuine Prusa Firmware on genuine Prusa Hardware because pre-set by "us".
Fun fact. If you were to re-calibrate the Extruder stepper motor on genuine Prusa hardware and firmware you may notice a more or less 6% under-extrusion. That is not concerning: it works like that for legacy reasons, so almost intentional and we don't recommend re-calibrating.

Prusa Slicer is based on Slic3r engine.
If you want to know more about the magic behind it, please consult their website and GitHub.
https://slic3r.org
https://github.com/slic3r

If you have a specific question about Prusa Slicer instead, please post it in the relevant repository.
https://github.com/prusa3d/PrusaSlicer

Michele Moramarco
Prusa Research

Prusa-Support, 2022-11-29:
I assume this issue can be closed now.

Michele Moramarco
Prusa Research

closed 2022-11-29 by Prusa-Support