prusa3d/Prusa-Firmware #3735: [FEATURE REQUEST] Synch bed and nozzle heating

KilianB, 2022-11-05:
**Is your feature request related to a problem? Please describe.**

When printing with a high print bed temperature heating up the bed usually takes a lot longer than heating up the nozzle. During this period filament might start to ooze from the nozzle and some types of filaments loose their properties if exposed to high temperature for an extended period of time.

Here is an example with ASA where the nozzle hit 260° and is sitting idle for a few minutes until the bed reached its target temperature.

[image]

**Describe the solution you'd like**

The hotend and bed should reach their maximum temperature at roughly the same time.

**Describe how it would work**

With the new temperature calibration feature rolled out in the current beta firmware we are aware of expected temperature profiles for a printer. Using this we could defer heating either the bed or nozzle until the appropriate point in time.

A more trivial approach would be to gradually increase or decrease the power of the heating element depending on the temp delta.

Ref: https://github.com/prusa3d/PrusaSlicer/issues/9127

adelyser, 2022-11-05:
Why would you implement this in firmware when you have the flexibility to make this happen in your start GCODE? Firmware shouldn't be overly complex to handle things that can easily be scripted from the slicer, IMHO. Here's how I handle it for no-ooze....

```
M104 S160 ; set extruder temp to pre-heat
M190 S[first_layer_bed_temperature] ; wait for bed temp

G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

M109 S[first_layer_temperature] ; wait for extruder temp
```

KilianB, 2022-11-05:
Maybe the issue should be moved back to the slicer. I just feel that it is a reasonable default and the firmware can be a bit smarter about it than just playing around with the heating order.

Prusa-Support, 2022-11-25:
Thank you for sharing your idea.

I find it reasonable to keep the preheating process simple as it is now, so the nozzle can ooze a little before the start of the print, otherwise it may slightly ooze during the initial calibration.

The little void in the nozzle created by oozing will be compensated for, during the "intro purge line", where also the plastic (possibly mildly degraded by heat) is purged.

In other words, the suggested idea may not necessarily be beneficial. It may seem to be a personal preference at this point, and Prusa Slicer would allow individuals to set up a custom start g-code.

Michele Moramarco
Prusa Research

closed 2023-11-09 by github-actions[bot]