prusa3d/Prusa-Firmware #3738: [Community made] Romanian language translation fix

vrosu, 2022-11-06:
Prusa Research will NOT follow up these issues!
The maintainers of the "Community made" feature should/will react.

Please, before you create a new "Community made" ticket, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Which Community made feature do you want to address?**
**Romanian language translation.**
@leptun @Hauzman 

**What is your request/question/suggestion?**

1. Message "MSG_WIZARD_WELCOME" is translated as "Salut, sunt imprimanta ta Original Prusa i3. Vrei sa initiez testele automate si calibrarile?".

Even if it is clearly in the spirit of it, the existing translation is not a 1:1 translation of the english message, 
This should be "Salut, sunt imprimanta ta Original Prusa i3. Vrei sa te ghidez prin procesul de instalare?"

2. Message "MSG_LAST_PRINT_FAILURES" is translated as "Err. ultimului print". It should be "Erorile ultimului print".
3. Message "MSG_PREHEATING_TO_LOAD" is translated as "Preincalzire load". It should be "Preincalzire incarcare"
4. Message "MSG_PREHEATING_TO_UNLOAD" is translated as "Preincalzire unload". It should be "Preincalzire scoatere"
5. Message "MSG_SEVERE_SKEW" is translated as "Dist. severa". It should be "Deformare severa"

3d-gussner, 2022-11-07:
@vrosu Thanks for your report.
The translators should have a look at the 1st message but the 2-5 are okay, as the messages have length limits and your suggestions exceed these limits. 

- `MSG_LAST_PRINT_FAILURES` c=20 which means max 20 chars allowed
- `MSG_PREHEATING_TO_LOAD` c=20
- `MSG_PREHEATING_TO_UNLOAD` c=20
- `MSG_SEVERE_SKEW` c=14

vrosu, 2022-11-07:
@3d-gussner am I allowed to make other suggestions for 2-5 in comments to this ticket or should I open another ticket?

3d-gussner, 2022-11-07:
Sure as long the don't exceed the limits.

vrosu, 2022-11-07:
I will do that at the end of today. Thanks

vrosu, 2022-11-08:
@3d-gussner @Hauzman  I modified the translations in the original post according to the max length.

Hauzman, 2022-11-08:
@vrosu I can have a look, but this was @leptun initial translation.
I will check it and change it. Also, I have some optimization

closed 2023-09-01 by 3d-gussner