prusa3d/Prusa-Firmware #3740: [BUG] 3.12.0-BETA1 Firmware printer resetting itself during 16+ hour prints via Octoprint

MajorGamerGeek, 2022-11-06:
MK3S - 3.12.0-BETA1 - USB/Octoprint

Description:
Printer resetting itself canceling the print when printing 16+ hours via Octoprint. This has happened twice to me during long 3D prints using .12.0-BETA1 via USB/Octoprint and is not an 100% repro.
  
To Reproduce:
1.) Connect Prusa i3 MK3S in a Enclosure with firmware 3.12.0-BETA1 to Octoprint version 1.8.6 via USB.
2.) Start attached GCode for 0.6mm Nozzle PETG Filament
3.) Wait 16+ hours and the printer will reset itself
  
Expected behavior:
3D Print completes successfully without the Printer Resetting itself.
[space_walkie_0.6n_0.2mm_PETG_MK3S_21h8m.zip](https://github.com/prusa3d/Prusa-Firmware/files/9946918/space_walkie_0.6n_0.2mm_PETG_MK3S_21h8m.zip)

Prusa-Support, 2022-11-07:
Hello, thank you for the report. Would it reset always after exactly 16 hours? Would another model possibly reset before, or would you notice an increase on the frequency of the resets? Would it reset also without Octoprint, or with older FW? I am not certain at this stage that it might be FW-related, there could be hardware causes. Also, do you possibly have a serial log or the message from OctoPrint to try and understand what the error was shown as in Octoprint? 

Regarding the GCode, unfortunately I can't download a compressed file due to security reasons, can you please send the GCode uncompressed? 

I await your update, we'll look into this and narrow down the causes! 

Alessandro Pantaleo
Prusa Research

MajorGamerGeek, 2022-11-07:
The was reset was about 16 hours into an 21 hour print. The printer would not reset on older release firmware via the same Octoprint Setup. Serial Logging within Octoprint was disabled at the time this occurred. I have enabled Serial Logging now for future prints and issues occurred. This has happen to me twice with long prints on MK3S 3.12.0-BETA1 Firmware with different GCode. I was next to the printer once when it happened and could confirm the printer reset.

GitHub only allows me to upload files of a Maximum file size of 25MB so I cannot attach the uncompressed CGode. If you have a location I can upload the file to let me know.

Regards,
Adam

arekm, 2022-11-07:
There should be something in octoprint.log (or log with date) anyway.

closed 2023-11-09 by github-actions[bot]