prusa3d/Prusa-Firmware #3742: [BUG] 3.12.0-BETA1 + MMU2S - printer is stuck on MMU2S failure - it does not react to MMU2S buttons or recovery (e.g., loading filament)

RPospisil, 2022-11-08:
**Printer type** -MK3S+
**Printer firmware version** -  3.12.0-BETA1

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**USB/Octoprint**

**Describe the bug**
When MMU2S is in failure state (e.g., filamant is not loaded, or out-of-filament), then printer displays error message containing like "press button on MMU unit to continue". Anyway pressing any button on MMU2S unit or recovering the MMU2S failure (by loading the filament) is not read out by printer and printer is stuck forever, so only possible solution is to reset it.   
  
**To Reproduce**
Use printer menu to load filament but do not insert the filament to MMU2S. After a while printer displays error message and it is not possible to recover printer by loading the filament (using MMU2S buttons). Printer is stuck and the only solution is to reset printer.
  
**Expected behavior**
When MMU2S failure is resolved (e.g., by loading the filament using MMU2S buttons) printer shall continue.

**G-code**
N/A

**Video**
NONE

ursius2020, 2022-11-14:
I can confirm and also found out this problem is present since v3.10
Also talked with the Support about this…
I hope this issue will be resolved soon or with v3.13 when also the MMU firmware is reworked

E-Extrude, 2022-11-15:
Another confirmation.
Fully stock MK3S and MMU2S.
Once MMU issue is resolved the MMU responds to the buttons as if it isn't mid print and the main knob on the printer just cycles the MMU error message. No option but to reset printer  and scrap print :(

Prusa-Support, 2022-11-29:
Thanks for bringing this to our attention.
Have you actually tried resetting the MM-control board by pressing the embedded button?
Use a hex key to reach it from the right side, it can be found next to the micro-USB port. At that point, the Multi Material should reboot and resume the previous operation, without losing the print progress.

Michele Moramarco
Prusa Research

E-Extrude, 2022-11-30:
When the MMU reset button is pushed, the unit homes the idler and selector barrel (unsure on correct term) and a green light appears above the filament selected. Just as you would hope, all errors have been cleared and the MMU is now ready. But the message isn't getting to the printer to resume printing.
Thanks for your time looking into this.

RPospisil, 2022-12-02:
I've re-tested the issue again on 3.12.0-RC1 and it is still there. AFAIR the last version where it worked was 3.9.x .

@Prusa-Support I've tried to reset the MMU2S unit as you suggested - it did not help (personally I did not believe that it is linked with the issue).

Prusa-Support, 2022-12-07:
Thank you for clarifying.
If I understand correctly the problem is a missing option on the User Interface to resume the print in this specific scenario. Probably this is not only a bug but a UI Feature Request. 

Our developers will look into this.

Michele Moramarco
Prusa Research

RPospisil, 2022-12-07:
Hi, 
IMO it is defect/bug. It is not feature request because it worked before (AFAIR ~3.9.x). For example (see previous posts) during the failed filament loading (executed from the printer menu "load filament") following errors are displayed on the printer:

#~ msgid "MMU needs user attention."
#~ msgstr "MMU potrebuje zasah uzivatele."

#~ msgid "Fix the issue and then press button on MMU unit."
#~ msgstr "Opravte chybu a pote stisknete tlacitko na jednotce MMU."
 
In the past once I loaded the filament using MMU2S buttons, the printer reads the status from the MMU2S and continued. In the current firmware I have to do reset the print.

Further the communication between MMU2S and printer is broken and for example a simple recoverable failure (such as failed loading the filament on filament switch) ends with necessity of the printer reset (but it worked in past)...

Hauzman, 2022-12-09:
I can confirm to when the MMU is not working also on the 3.12.0-RC1. same think on my MK3S+

Mirarkitty, 2022-12-17:
Can confirm. Issue in beta1 and rc 1 for me. Hoping for resolution before release...

gudnimg, 2022-12-17:
Can anyone confirm which firmware this issue first appears in? 3.10.0? 3.10.1?

It seems like this issue isnt new with 3.12. 

If this issue has been present for more almost 2 years, I wouldnt be surprised if Prusa Research decides to focus on 3.13 instead. Currently this isnt an issue with the upcoming MMU FW.

ursius2020, 2022-12-17:
The Problem started with 3.11

Starting with 3.10.0 OctoPrint PSUControl is no longer turning off the PSU after the print is finished and cooled down. So i am now using 3.9.3 until 3.13 comes out and hopefully all issues have beed resolved

Prusa-Support, 2022-12-17:
Our developers have been working on the total rewriting of the MM-control-board FW to enable more responsive and user-friendly interaction with the user via a printer LCD panel. They are still working hard on it and probably reworking on the Einsy FW side too.

Thank you for your feedback, our developers will look into this.

For Octoprint-related problems though, it may take extra time as it is third-party software.

Michele Moramarco
Prusa Research

RPospisil, 2022-12-17:
@Prusa-Support thanks for information. If devs will confirm this as a bug, will it be fixed in upcoming release? What's the expected release/time of new MM + MK3S firmware?

DRracer, 2022-12-20:
The new MMU FW is being intensively tested and polished along with the upcoming FW 3.13.
FW 3.12 "only" brings the thermal model checks which is an huge improvement in printer's safety on its own.
Originally, we were to release both features (thermal model + new MMU implementation) at once but due to severe delays in the new MMU testing we decided to separate both releases.

For now, please stick with FW 3.9.x, where it still works (based on your reports), thank you for your understanding. We confirm this is a long standing bug, however attempts to fixing it may seriously break something else (shortly before releasing 3.12).

3dprintandpaint, 2023-01-12:
Also having the same issue. I seem to be having frequent MMU failures, which were not present before. This could be coincidence. As described above, when the MMU issue triggers, there is no going back and the print is lost. 5 x attempts have all failed to date.

3d-gussner, 2023-01-13:
@RPospisil Can you please check this firmware as well? See https://github.com/prusa3d/Prusa-Firmware/issues/3895#issuecomment-1382105362

RPospisil, 2023-01-13:
@3d-gussner Hi, I've tested the fw 5673 build (from the comment  https://github.com/prusa3d/Prusa-Firmware/issues/3895#issuecomment-1382105362 ) in following usecases

- loading the filament
I've run action "load filament 1" from the menu on the empty slot. After a while printer displayed error message that "I have to resolve the issue and press a key on the MMU unit". So I just press a button on the MMU unit and printer returned back to the load filament menu.

- out-of-filament in the middle of the print
When printer detected out-of-filament, then I loaded a new filament to the same slot (I was able to change the slot too) and after pressing button on the printer, the filament was loaded into the nozzle, I selected, that the loading is ok, and the print successfully continued.

Summary: It work, but it seems (AFAIR) that it work little bit differently comparing to the latest working firmware (AFAIR ~3.9.x).
I did not tested stuck filament or other situations, so I cannot comment on them, but I believe, that it should work too.

@3d-gussner thanks! Are you going to merge your changes and release 3.12.1 :-) ? Or - as @DRracer mentioned before - we have to wait for a new FW 3.13 + new MMU FW?

regards, -rp-

3d-gussner, 2023-01-16:
@ursius2020 @Mirarkitty @Hauzman Would be great if you could also test this firmware and give us feedback.

The fix has been merged for FW3.12.1.

Hauzman, 2023-01-16:
@3d-gussner will test it tomorrow when i get back and let you know

3d-gussner, 2023-01-16:
> @3d-gussner will test it tomorrow when i get back and let you know

Thanks @Hauzman

Hauzman, 2023-01-17:
@3d-gussner Hi,

I can confirm the error, if in the mmu is not inserted the filamentet or if it's a loading problem the only whay is the hard reset. This was also in the previous version to.
I have attached picture one from the mmu with the green LED and the other one is from the LCD

3d-gussner, 2023-01-17:
Please try fw 3.12.1—RC1

Hauzman, 2023-01-18:
> Please try fw 3.12.1—RC1 

@3d-gussner I have tested also the latest firmware 3.12.0 and also 3.12.0 RC1 the same thing was for me.
And I have put a 4.5-hour print( FW 3.12.0) last night with color change, and at the finish and when the filament retracted, it gives the same error, but the print this times was finished.

I WILL test today the fw 3.12.1 RC1

3d-gussner, 2023-01-18:
Easy test procedure
- don't put filament in the mmu2
- choose lcd load filament - > load filament 1
- wait for error
- push filament into selector so FINDA triggers
  - it should resume
- put filament 2 in mmu2s
- load filament 2

ursius2020, 2023-01-18:
I‘ve tested it via the easy test procedure, issue is gone.
Haven’t tested if every thing works fine while printing but should work…

Hauzman, 2023-01-20:
> Easy test procedure
> - don't put filament in the mmu2
> - choose lcd load filament - > load filament 1
> - wait for error
> - push filament into selector so FINDA triggers
>   - it should resume
> - put filament 2 in mmu2s
> - load filament 2

@3d-gussner i can confirm to it's working the error is gone when i put the filament and it load the lcd i back to normal to load filament

Prusa-Support, 2023-01-25:
> **Non responding MMU2S fixed**
> Firmware 3.12.0 accidentally introduced a bug, where the MMU2S unit might stop responding in case of an error. For more details, see issue https://github.com/prusa3d/Prusa-Firmware/issues/3742. This is now fixed and the unit will work as expected.
> 
> We would like to thank our community members @RPospisil, @ursius2020, @E-Extrude, @Hauzman, @Mirarkitty and @3dprintandpaint for reporting this bug and retesting the fix.

https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.1

I hope you don't mind me closing this issue as solved. However, please feel free to share further feedback on this thread.

Michele Moramarco
Prusa Research

closed 2023-01-25 by Prusa-Support