prusa3d/Prusa-Firmware #3744: [BUG] mk2.5S hidden "experimental" menu is empty

filewalker1, 2022-11-09:
**Printer type** MK2.5S
**Printer firmware version** -  3.10.0, 3.10.1, 3.11.0, 3.12.0-Beta1

**Describe the bug**
  It is possible to activate the hidden "experimental" Menu in the HW-Settings, but the menu is empty, no options at all.
  
**Expected behavior**
  see a similar menu as in the MK3, maybe missing the options which are mk3 specifig (e.g. e-cool mode - i think this feature is not available with the old stepper drivers?).

leptun, 2022-11-10:
I'm not sure if anything could be placed in the menu. The available options are MK3 specific. We could hide the menu in the MK2.5 builds, but it's not like the MK2.5 is tight on resources at the moment.
@gudnimg Any idea what we could put here?

filewalker1, 2022-11-11:
okay, i was not aware that none of the option is supported by mk2.5. If this is the case it is totally fine for me to leave this menu in MK2.5 builds. I am having problems with SNR recognition and support told me to try fake SNR from experimental menu, so i thought i is maybe a bug, that this option is not available on MK2.5

gudnimg, 2022-11-11:
> I'm not sure if anything could be placed in the menu. The available options are MK3 specific. We could hide the menu in the MK2.5 builds, but it's not like the MK2.5 is tight on resources at the moment. @gudnimg Any idea what we could put here?

I think we can leave the menu empty, similarly how there is an empty menu when there is no SD card connected. I don't really see this as an issue.

closed 2023-11-13 by github-actions[bot]

3d-gussner, 2025-03-20:
@mrecman Changing the EEPROM value isn't enough as the variants for the MK2.5/S don't have
```
#define EXTRUDER_ALTFAN_DETECT
#define EXTRUDER_ALTFAN_SPEED_SILENT 128
```

You can change `#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed` to `#define EXTRUDER_AUTO_FAN_SPEED   128  // == alt fan speed` and compile the firmware.

>Please return menu to MK2.5S firmware.

In which version was it available?

mrecman, 2025-03-25:
> [@mrecman](https://github.com/mrecman) Changing the EEPROM value isn't enough as the variants for the MK2.5/S don't have
>
> ```
> #define EXTRUDER_ALTFAN_DETECT
> #define EXTRUDER_ALTFAN_SPEED_SILENT 128 [...]

Sorry. I mean ADD EXPERIMENTAL menu to MK2.5S firmware.

But I try compile firmware myself. Thanks.