prusa3d/Prusa-Firmware #3751: [BUG] Sending either (or both) G98 / PRUSA Firm commands will interrupt ongoing print.

bolsoncerrado, 2022-11-12:
MK3S
3.11.0
No MMU
USB Printing with Octoprint

**Describe the bug**
Printer is printing a job.
I go to Octoprint Terminal, issue a G98 and shortly after a PRUSA Firm
Job stops, connection is lost.
  
**To Reproduce**
See above
  
**Expected behavior**
Firmware reported OK, it's just I find odd the print job is lost, runied and connection lost.

  
**G-code**
[...]
Send: N59857 G1 X89.931 Y152.383 E8.2226*108
Recv: ok
Send: N59858 G1 X89.815 Y152.644 E8.2329*100
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Recv: ok
Send: N59859 PRUSA Firm*3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409*111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
Printer keeps requesting line 59859 again and again, communication stuck
Changing monitoring state from "Printing" to "Error"
Send: M112
Send: N59861 M112*34
Send: N59862 M104 T0 S0*33
Send: N59863 M140 S0*100
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
Closing down send loop

gudnimg, 2022-11-12:
The firmware looks for the command : `PRUSA Fir` not `PRUSA Firm`. Does `PRUSA Fir` also lead to the same result?

gudnimg, 2022-11-12:
From looking into the code:

```
Send: N59859 PRUSA Firm3
Recv: 3.11.0-4955
Recv: ok
Send: N59860 G1 X89.675 Y152.821 E8.2409111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
Recv: Resend: 59859
```

In the first line here `Send: N59859 PRUSA Firm3`, If `"PRUSA"` is in the same serial message as the requesting new line command `"N"`, then the line request will be skipped/ignored.

This is why you get an error in the next `"N"` command. Because line `59859` was never actually processed.
```
Send: N59860 G1 X89.675 Y152.821 E8.2409111
Recv: Error:Line Number is not Last Line Number+1, Last Line: 59858
```

gudnimg, 2022-11-12:
Apparently "PRUSA" hasn't been allowed with "N" command for a long time. Since MK2: https://github.com/prusa3d/Prusa-Firmware/commit/e8477b525c1df1e25ccaf3487cdebc2a1e881beb 5 years ago at least

bolsoncerrado, 2022-11-13:
Not fully sure what you mean. I type the G98 and PRUSA Firm command from the Terminal window on Octoprint while the print was ongoing. Did this mess things up?

Anyhow seems PRUSA FARM command G98 stops any USB activity with Octoprint so in the end it's either a no-go feature or another bug... :?

gudnimg, 2022-11-13:
I suspect G98 is different. I don't see any reason why G98 will fail. What does the Octoprint error look like in that case?

bolsoncerrado, 2022-11-13:
There's no fail other than the whole disconnection. While FARM mode is still enabled, Octoprint won't connect (just times out trying)

bolsoncerrado, 2022-11-13:
Meaning G98 actually suceeds.

gudnimg, 2022-11-13:
> Not fully sure what you mean. I type the G98 and PRUSA Firm command from the Terminal window on Octoprint while the print was ongoing. Did this mess things up?

To answer you, I believe `PRUSA Firm` is incorrect, it should be `PRUSA Fir`. But you will likely still face the same problem because... as far as I can understand the firmware code **N** and **PRUSA** (the texts) cannot be in the same string.

Send: **N**59860 **PRUSA** Fir*3

The firmware will interpret/parse this line like this: it sees `"PRUSA"` and from that alone it will ignore everything including N59860. This is important because the firmware needs to read the number 59860 and store it in memory, but it doesn't do that when PRUSA text is seen.

> Anyhow seems PRUSA FARM command G98 stops any USB activity with Octoprint so in the end it's either a no-go feature or another bug... :?

Maybe @Prusa-Support can help answer this :) I'm not sure if Prusa's print farm uses Octoprint.

leptun, 2022-11-13:
The problem is that all these text based commands may contain the N character, so line number detection might fail when no line number is provided. I think this is why line mumbers are prohibited for PRUSA commands. The only realistic solution would be to make sure the N position (from code_seen) is before "PRUSA...".
code_seen searches for the first character in the string with that letter (N in this case).

bolsoncerrado, 2022-11-13:
So the issue is how Octoprint sends terminal commands from terminal text box?

leptun, 2022-11-13:
No. It's the firmware's fault. We need to handle line numbers better.

wavexx, 2022-12-12:
>  I think this is why line mumbers are prohibited for PRUSA commands. 

:vomiting_face: 

> The only realistic solution would be to make sure the N position (from code_seen) is before "PRUSA...".
> code_seen searches for the first character in the string with that letter (N in this case).

Aren't line numbers _required_ to be as the first item in the gcode line (just like checksum is last)?

leptun, 2022-12-12:
Yes, they are supposed to be the first thing in the line, but it's not like that is checked anywhere, or is it?

wavexx, 2022-12-12:
If it isn't checked, we should definitely add this.

wavexx, 2022-12-14:
> Apparently "PRUSA" hasn't been allowed with "N" command for a long time. Since MK2: [e8477b5](https://github.com/prusa3d/Prusa-Firmware/commit/e8477b525c1df1e25ccaf3487cdebc2a1e881beb) 5 years ago at least

I had to look at this again for another issue. #3823 should fix this and many related issues.

As a summary:

- PRUSA Firm is indeed wrong, although it will parse fine
- We now process line numbers early, so PRUSA is recognized with numbers and checksums as well
- The parsing is stricter, so if the command is not recognized it will now error correctly instead of being silently dropped.

N is indeed *already* expected to be first after a newline. No whitespace is allowed before N, so N cannot be parsed in the middle unless there's line noise. But since N *expects* a checksum, we should see this.

We no longer allow whitespace before the gcode is seen. I wonder if this has *any* impact, since it was only possible with SD printing. Please comment on #3823 on this.

We do allow to mix commands with and without line numbers in the stream. I see this as a fault of the protocol we probably can't do anything about...

leptun, 2022-12-14:
@wavexx The only fault I can see with this is what would happen with a gcode file that was uploaded to the SD over usb. I'm pretty sure in the past the line numbers would get saved to the file on the SD, but with $ instead of N. If that is the case, then all lines will be parsed incorrectly. I played a few months ago with this, but I don't remember if I made it so it doesn't save those line numbers to SD. Can you try uploading a file to the SD and check how it looks like?

closed 2022-12-30 by DRracer