prusa3d/Prusa-Firmware #3753: [BUG] thermal model protection triggers when preheating

ksipple, 2022-11-15:
I got a thermal anomaly error when preheating (PLA) in a cold room (16C), I calibrated when the room was slightly warmer.

    Reproducible: Yes
    Printer model: MK3S+ 
    Modifications: E3D Revo, X2 Delta P Duct [V2R2.2] Fan shroud (https://www.printables.com/model/155387-remix-delta-p-duct-v2r2-for-the-mk3s-x2-duct-new-f)

LOG
-------
Recv: T:28.9 /0.0 B:59.1 /0.0 T0:28.9 /0.0 @:0 B@:0 P:0.0 A:28.6
Recv: MMU not responding - DISABLED
Recv: T:29.2 /0.0 B:59.1 /0.0 T0:29.2 /0.0 @:0 B@:0 P:0.0 A:28.6
Recv: T:28.4 /0.0 B:58.8 /0.0 T0:28.4 /0.0 @:0 B@:0 P:0.0 A:28.4
Recv: tmc2130_home_enter(axes_mask=0x04)
Recv: T:28.3 /215.0 B:58.8 /60.0 T0:28.3 /215.0 @:127 B@:127 P:0.0 A:27.9
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
Recv: T:31.1 /215.0 B:58.6 /60.0 T0:31.1 /215.0 @:127 B@:127 P:0.0 A:28.5
Recv: TM: error |1.400330|>1.200000
Recv: LCD status changed
Recv: T:39.6 /215.0 B:58.4 /60.0 T0:39.6 /215.0 @:127 B@:127 P:0.0 A:28.2
Recv: TM: error |1.471889|>1.200000
Recv: LCD status changed
Recv: TM: error |1.527903|>1.200000
Recv: LCD status changed
Recv: TM: error |1.612518|>1.200000
Recv: LCD status changed
Recv: TM: error |1.681316|>1.200000
Recv: LCD status changed
Recv: TM: error |1.747072|>1.200000

ksipple, 2022-11-15:
I did another calibration just now (room is still cold) and then tried to preheat and got another thermal anomaly error. Also, to add I did a successful print when the room was warm yesterday.

LOG
-----
Recv: T:123.2 /215.0 B:26.7 /60.0 T0:123.2 /215.0 @:127 B@:127 P:0.0 A:27.6
Recv: TM: error |1.445011|>1.200000
Recv: LCD status changed
Recv: TM: error |1.429280|>1.200000
Recv: LCD status changed
Recv: TM: error |1.379950|>1.200000
Recv: LCD status changed
Recv: TM: error |1.351405|>1.200000
Recv: LCD status changed
Recv: TM: error |1.309368|>1.200000
Recv: LCD status changed
Recv: TM: error |1.265119|>1.200000
Recv: LCD status changed
Recv: TM: error |1.217088|>1.200000
Recv: LCD status changed
Recv: T:129.8 /215.0 B:27.1 /60.0 T0:129.8 /215.0 @:127 B@:127 P:0.0 A:27.7
Recv: TM: error |1.217088|>1.200000
Recv: T:136.2 /215.0 B:27.9 /60.0 T0:136.2 /215.0 @:127 B@:127 P:0.0 A:27.6

darraghbr, 2022-11-17:
It's because of the Revo you have installed, this is a duplicate of #3636

madproducts, 2022-11-24:
> It's because of the Revo you have installed, this is a duplicate of #3636

Is here any resolution in the pipeline to either this bug or #3636 ? Or will Revo not be supported at all?

darraghbr, 2022-11-24:
They seemed keen to capture the behaviour of the Revo, but ultimately it's early days for the concept as a whole so who knows.

madproducts, 2022-11-24:
> They seemed keen to capture the behaviour of the Revo, but ultimately it's early days for the concept as a whole so who knows.

Agreed - For now, post calibration M310 S0 to disable Thermal model and then M500 (save) seems like the only way forward. Not going to change my Revo for Tcal - that's for sure.

Prusa-Support, 2022-11-29:
Thanks for reaching us. Please share your concerns and findings in the existing issue #3636.
I hope you don't mind me closing this one as a duplicate.

Michele Moramarco
Prusa Researech

closed 2022-11-29 by Prusa-Support