prusa3d/Prusa-Firmware #3764: [BUG] MMU2S motor does not change position after soft reset or do not print at all

sramekmichal, 2022-11-19:
**Printer type** - MK3S 
**Printer firmware version** -  3.11.0

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**SD card or USB/Octoprint**
BOTH

**Describe the bug**
I got fixed my printer from Prusa and sometimes during print I get an error with all lights blinking. To continue, I need to solve a MMU2S problem and then click a button. After clicking on each button, the light still blink. Everytime only soft reset do the trick, but the print is ruined, because the motor does not work and do not change position. It remains on the same place whatever I am trying to do. I did not change anything since I got it from Prusa service.
  
**Expected behavior**
 To continue printing a.k.a. changing the filament slot.
 
 MMU2S is really PAIN!

Prusa-Support, 2022-11-24:
Hello, thank you for the report, there might be several causes behind this - can you please reproduce the issue and send me some videos via email info@prusa3d.com, mentioning me? Can you also please send me via email the order number of the MMU unit? That way I can see the service history and we can review everything in more detail. Thank you. 

Alessandro Pantaleo
Prusa Research

sramekmichal, 2022-12-20:
Hi, I tried to fix it first by myself...I am kinda desperate, nothing helped :-( 
I recorded video and with deep description I send an e-mail.

Appreciate your help so much! It became hell unit...
Thank you very much.

closed 2023-11-13 by github-actions[bot]