prusa3d/Prusa-Firmware #3765: [BUG] Could not open /dev/cu.usbmodem141201 (MMU2S)

sramekmichal, 2022-11-19:
**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6. (allegedly, seen in printer)
**SD card or USB/Octoprint**
BOTH

**Describe the bug**
Following issue #3764 I tried to solve a problem with MMU2S unit by hard reset. Further I wanted to re-flash the firmware (on Mac, with PrusaSlicer), but connection to the port is getting timeout and `could not open /dev/cu.usbmodem141201`.

How can I flash firmware or open the port?
  
**Expected behavior**
Be able to flash MMU2S unit by original usb cable.

Prusa-Support, 2022-11-24:
Hello, thank you for the report. This type of issue is typically resolved, on Mac-Linux, by either using **sudo** command in Terminal or right-click on the application and **Run as Administrator**. On another hand, can you please double-check if 

For the sudo command, it should be resolved this way: 
1. Open terminal app
2. Insert: sudo ./PrusaSlicer-(...).AppImage

For the other issue #3764 it would be best if we looked into it via email, so that we can share data from the printer order and pics-videos over email, I replied there with more details. 

Alessandro Pantaleo
Prusa Research

sramekmichal, 2022-11-27:
Thank you, it works.

closed 2022-11-27 by sramekmichal