prusa3d/Prusa-Firmware #3778: [FEATURE REQUEST] Option to disable safety timer

Quinten0508, 2022-11-23:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - Mini+ (but most likely other Prusa printers too)

**Is your feature request related to a problem? Please describe.**
I use Octoprint for most of my prints, and pause longer prints during the night. However, if I pause and then unpause a print (using the Octoprint UI), the FW's safety timer will kick in. After 30mins of printing, both the nozzle and the bed heating turn off, but Octoprint still sends GCODE to the printer and the entire thing keeps "printing". It just spits out temperature errors since it cannot extrude anything with a room-temperature nozzle. This effectively ruins any print, and since this only applies to long prints, you can imagine the level of frustration caused when a perfectly fine print is turned into waste due to a FW issue.
As a workaround I no longer pause the prints but cancel them exactly after a layer is done, then I remove all the GCODE it has printed thus far and print this new file - essentially continuing the print where it left off. However, this does **not** trigger the safety timer.
  
**Describe the solution you'd like**
A simple on/off toggle in the settings which allows for someone to turn off the safety timer. You could add a big red warning screen and turn the timer back on after every boot in order to ensure safety.

Quinten0508, 2022-11-23:
> I do not agree that a gcode to disable the safety timer is the right way, but what you describe here:

Not necessarily I was referring to, a toggle in the printer's UI was more what I envisioned. A special GCODE command also does the job, but needs an additional warning upon printing and an option to ignore this command pre-print in case a GCODE file is shared on e.g. Printables with this GCODE inside.

> Are both the pause and unpause through Octoprint? (Printer’s LCD not involved?)

Correct. I haven't done a whole lot of testing yet as a 3-day print (using the workaround) finished just a few hours ago. It is definitely worthwhile to try pausing and/or unpausing via the LCD to rule out either Octoprint or the FW. 
Same goes for anyone else with a Prusa printer and Octoprint; getting more info on what printers are affected is crucial here.

Octoprint is fully up-to-date, and I confirmed no plugins were causing the problem. My Mini+ is running on the latest stable release, no betas have been tested. I'll see if I can get replicate the issue with the Beta version you mentioned, but I'll be leaving for a conference tomorrow until Sunday, so any other testers are welcome 😅

The safety timer is _very_ important and we don't want houses to burn down because there were no clear indications that the timer was turned off. It's much better to patch whatever bug is causing this but having an override is useful regardless.

gudnimg, 2022-11-23:
Since you have a Mini, it might be better to open an issue here: https://github.com/prusa3d/Prusa-Firmware-Buddy as the firmware on the Mini and MK3/S are totally different. 

I can’t test your issue on my MK3S+ unfortunately since I don’t have a raspberry pi at the moment (been waiting for months for it to come back into stock 👀). But I have forwarded you bug report into Prusa’s internal Slack, so maybe someone can help reproduce the issue to confirm.