prusa3d/Prusa-Firmware #3785: [BUG]<Enter comprehensive title>

JorLars, 2022-11-27:
**Printer type** - MK3S+ no mods other than silicone sock
**Printer firmware version** -  3.12.0.RC1

**MMU upgrade** - No
**MMU upgrade firmware version** - No

**SD card or USB/Octoprint**
SD

**Describe the bug**
  Directly after Temp model cal. finished, the printer cooled down from 230 to aprox 120 degrees C when Thermal Anomaly error appeared and wouldnt stop beeping until reset. And when it had reset it started beeping again. 
I turned off printer because of the loud beep. When cooled I powered on again
Tried a print and when temp came up to around 55 degrees thermal anomaly appeared again.

Ambient temp is 21 degrees C it's in a Lack enclosure.
  
**To Reproduce**
Any print
  
**Expected behavior**
No thermal anomaly
  
**G-code**

**Video**

JorLars, 2022-11-28:
Tried Temp model cal. with printer out of enclosure and got the same result as above.
Had to roll back FW to 3.11 to be able to start printing.

QR4rigs, 2022-12-01:
I get the same on 1 of my Mk3S+ printers that has a Revo six hotend. Any upmost down temp request leads to C a Thermal anomaly. The other identical Mk3S+ printer does not suffer with this issue

Prusa-Support, 2022-12-03:
Thanks for reaching us.

Silicone socket and enclosure should not be a problem but may add complexity that requires tweakings on some parameters sometimes.

The thermal model calibration should possibly be repeated every time the ambient and hardware are modified in any way but it may not be all.
In the case of enclosures where the temperature is not actively controlled, the initial ambient conditions could vary as the enclosure heats up. Also, in the case of venting fans, make sure these don't generate airflows on the nozzle or drastic alterations of the ambient conditions.
The Einsy board temperature is used as a proxy. When the board is placed outside of the enclosure, you may need to adjust a temperature offset parameter. 
In the case of silicon socket or enclosure (or a combination of the two), the warning threshold may need to be modified.

Make sure that hotend fan and print fan RPM are in the range, and correctly aiming.
Make sure that thermistor and heater are in a good state and within the specifics.
Also make sure the ambient thermistor at the bottom of the Einsy board is in a good state.

Please find all details about the thermal model protection at https://github.com/prusa3d/Prusa-Firmware/pull/3552 and please mind:
>If the problem is more severe, the printer will warn with a strong, continuous beep and pause the print. All heaters are disabled in this case (both bed and nozzle). “PAUSED THERMAL ERROR” is shown.
>
>The thermal model itself cannot know what is going wrong! You shouldn’t ignore this!

Please take your time to review the Pull Request and make your considerations.
If you agree, this issue could possibly be closed.

.

Problems with Revo hotends should be reported in the specific issue.
https://github.com/prusa3d/Prusa-Firmware/issues/3636

Michele Moramarco
Prusa Research

wavexx, 2022-12-08:
@JorLars do you know how to log the serial output from the printer?

It would be nice if you could run the following commands:

```
M310
M155 S1 C3
D70 I1
M310 A
```

save *all* the serial output (starting from reset and until it's cooled down) and attach it here. Calibrating with a silicone sock shouldn't cause this issue.

QR4rigs, 2022-12-08:
@wavexx doyou need any more logs from Revo six? I have both the white and blue wired versions.

ognotongo, 2023-01-31:
I have a kit built (only been running a bit over a month) stock MK3S+, no mods, but printing via octoprint/usb.

I'm getting thermal anomaly errors frequently with 3.12.0 and 3.12.1. Zero issues before hand. I've tried thermal calibration, PID calibration and even a full run through of the wizard. Starts with frequent thermal anomaly and beeps, but usually progresses to an M112 emergency stop. Looking at the temperature graph in octoprint shows minor swings, 1-2 degrees Celsius, but nothing major. I will say that prior to the upgrade to 3.12.0, the temperature graph seemed much more stable, but I can't prove that. 

Also, another note, I've had the thermal anomaly trigger during the thermal calibration at least once. 

If they are still desired, I can try to generate the logs desired and attach them. It seems like it happens with every print now. Looking to do a factory reset soon so I can start printing again.

wavexx, 2023-01-31:
@ognotongo yes, please produce some logs. After this you can switch off the model directly with 3.12 via `M310 S0` + `M500` instead of downgrading the firmware.

ognotongo, 2023-02-01:
[PrintLogs.zip](https://github.com/prusa3d/Prusa-Firmware/files/10553510/PrintLogs.zip)

Here is a log from a failed print. Hopefully this can help. I wasn't sitting by the printer for this, but I never heard the beeping of the thermal anomaly warning, just the long tone of the emergency stop. I included a screen shot of the octoprint temp graph, though that can probably be more accurately recreated from the logs. And I also included the gcode that I was using for the print.

3d-gussner, 2023-02-01:
@ognotongo Thanks for the logfiles.

Looking at your logfile and the screen shot I would suggest that you run a LCD `Calibration -> PID calibration`  and again the LCD `Calibration -> Temp model cal.`

Your temperature is fluctuating quite a lot and overshoots during heat-up.

The error occurred on 1st layer which is very odd.
Do you have a stock print fan and stock fan shroud?
Please heat up your printer and "wiggle" your Extruder cable loom to check if there isn't a cable issue.
Check the hotend thermistor cable on your Einsy the connector pins.

Can you please try the same print with `M155 S1 C7 ;temp, fan and position autoreport` to check if it fails at same (close to it) spot.      

We don't fully understand why OctoPrint "thinks" that the printer send an error, as it should only react to `Error:` or `!!` there is a pull request to change it in next release. Meanwhile please try to disable OctoPrint to send `M112`
[image]

ognotongo, 2023-02-02:
My printer is completely stock and only about a month old. I did do the assembly myself.

Thermistor cable was fine and seated properly.
Wiggled the extruder cable bundle and didn't see any changes to anything, but wasn't sure what to look for. No temp changes and no warnings.

I ran the PID Calibration, Followed by the Temp Calibration, and finally the print. Logged it all (attached). Print again failed, and again didn't throw the thermal anomaly error. I didn't disable the M112 send as I wanted to reproduce the print from last night exactly. **Let me know if you want me to retest the print _with it disabled_**.

Interestingly enough, when preheating to wiggle the cable loom, I _did_ get the thermal anomaly. I'm going to try to capture it once more tonight before giving up. If I manage to, I'll post a log of that as well. I also noticed the temp reach 219 during the preheat before starting to go back down to the 215 target.

[serial_print.zip](https://github.com/prusa3d/Prusa-Firmware/files/10565615/serial_print.zip)

3d-gussner, 2023-02-02:
@ognotongo Thanks for your time to test things, but I think there is something wrong with your printer as you got a TM error `-4.7` at some point.

Please disable in OctoPrint the `M112` and try again a print with logging, we would like to see how often that happens and if that may be related to another part like Fan bursts.  

Normally your printer heats with `@60` to `@80` PWM with hotend at 208.2°C while target is 210.0°C but the last one is `@112`

@wavexx Do you have an idea what could cause such a TM error of `-4.7`?

At `@60`-ish heater PWM the print fan was `2400RPM` and `@84` pwm
At `@81`-ish heater PWM the print fan was `3840RPM` and `@170` pwm
At `@112`-ish heater PWM ther print fan was `4200RPM` and `@255` pwm
 
```
2023-02-01 21:14:59,640 - Recv: T:208.2 /210.0 B:60.0 /60.0 T0:208.2 /210.0 @:60 B@:6 P:0.0 A:33.7
2023-02-01 21:14:59,652 - Recv: X:140.05 Y:97.65 Z:0.35 E:37.86 Count X: 143.14 Y:98.18 Z:0.69 E:37.58
2023-02-01 21:14:59,661 - Recv: E0:4200 RPM PRN1:2400 RPM E0@:255 PRN1@:84
...
2023-02-01 21:17:38,222 - Recv: T:208.2 /210.0 B:59.9 /60.0 T0:208.2 /210.0 @:81 B@:78 P:0.0 A:34.5
2023-02-01 21:17:38,233 - Recv: X:129.75 Y:88.34 Z:0.50 E:19.55 Count X: 141.13 Y:99.36 Z:0.85 E:19.76
2023-02-01 21:17:38,244 - Recv: E0:4200 RPM PRN1:3840 RPM E0@:255 PRN1@:170
....
2023-02-01 21:36:48,410 - Recv: T:208.2 /210.0 B:59.9 /60.0 T0:208.2 /210.0 @:112 B@:61 P:0.0 A:37.2
2023-02-01 21:36:48,422 - Recv: X:133.29 Y:68.80 Z:1.55 E:3.59 Count X: 129.44 Y:83.83 Z:1.90 E:3.18
2023-02-01 21:36:48,429 - Recv: E0:4080 RPM PRN1:4200 RPM E0@:255 PRN1@:255
...
2023-02-01 21:36:48,794 - Send: N48367 G1 X150.143 Y61.212 E.03007*108
2023-02-01 21:36:48,816 - Recv: TM: error |-4.718834|>1.200000
2023-02-01 21:36:48,818 - Recv: LCD status changed
2023-02-01 21:36:48,849 - Recv: TM: error triggered!
2023-02-01 21:36:48,851 - Recv: LCD status changed
2023-02-01 21:36:48,877 - Recv: ok
2023-02-01 21:36:48,880 - Send: N48368 G1 X151.339 Y61.321 E.03129*97
2023-02-01 21:36:48,883 - Recv: echo:enqueing "G1 E-1.000 F2700"
2023-02-01 21:36:48,885 - Recv: // action:paused
2023-02-01 21:36:48,886 - Printer signalled that it paused, switching state...
2023-02-01 21:36:48,887 - Changing monitoring state from "Printing" to "Pausing"
2023-02-01 21:36:48,922 - Recv: Error:Printer stopped due to errors. Supervision required.
2023-02-01 21:36:48,924 - Changing monitoring state from "Pausing" to "Error"
2023-02-01 21:36:48,931 - Send: M112
2023-02-01 21:36:48,935 - Send: N48369 M112*33
2023-02-01 21:36:48,940 - Send: N48370 M104 T0 S0*41
2023-02-01 21:36:48,966 - Send: N48371 M140 S0*108
2023-02-01 21:36:48,986 - Changing monitoring state from "Error" to "Offline after error"
2023-02-01 21:36:49,004 - Connection closed, closing down monitor

ognotongo, 2023-02-03:
Here is the log requested. I had to switch to a new print, but I've included the gcode for that in the zip file. The printer had a hard time maintaining 215 towards the end of the print; seemed more stable earlier. In any event, I appreciate all of your help, but as this seems unrelated to the bug, I'll probably move this over to the standard support. 

I'm also going to downgrade to 3.11 and give it a shot. It could be coincidence, but that will tell me for sure. If the issues go away, I'll report back. 

Thanks again!

[serial_print_m112_disabled.zip](https://github.com/prusa3d/Prusa-Firmware/files/10575269/serial_print_m112_disabled.zip)

3d-gussner, 2023-02-03:
@ognotongo There are no `TM: errors` reported in this log file. Looking at the hotend temperatures they look good as the current temps have been within +-~0.5°C to the target temperature, even with Fan blowing at @255 = 100%

At the very end (finished) it is normal that the temperature spikes up as the fan stopped from 100% to 0% and the heater has some delay as it doesn't look in to the future gcodes.

If you don't mind please keep FW3.12.1 and OctoPrint with your current settings and in case of a `TM: error` try to recover using the LCD `Clear TM error`.

The last print was successful, but with the previous log I am afraid that somewhere you have an issue otherwise the TM wouldn't report such a big error value.

The whole idea of the thermal model is to detect way faster thermal issues.

Thanks a lot for testing and cooperation to our requests. Community members like you help us very much to improve the firmware. :hugs:

JorLars, 2023-02-18:
So I finally got my own stock MK3S+ and four other at a school working with 3.12.1.

I replaced heater thermistor and fans, took out heatbreak, cleaned everything, measured ohms of heater and thermistor, everything ok. Same problem. Just after upgrading to 3.12.1 the TM fails while heating up at about 96 degree C.

So just for kicks, I didn't place the heater evenly in the block, like the Prusa manual shows.
I put it through the block so you can just feel it on the pinda side of the heater block.
Now everything works. The TM goes through without any problems. Just did a few prints, no problems.
I also swapped back to my old fans, old thermistor and old heater. Works fine as long as I don't center the heater in the block.

If I put it like this it doesn't work.
https://cdn.help.prusa3d.com/wp-content/uploads/prusuki/prusuki-images/Dc4y4vR11YCMBUQU.jpg

Prusa-Support, 2023-03-03:
The recently released [FW 3.12.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.2) version improves the Thermal Model functionality. It also improves compatibility over serial connection and with some third-party hardware.

Please give it a try.

Michele Moramarco
Prusa Research

JorLars, 2023-03-06:
With 3.12.2 everything seems to be fixed.
I haven't had a single Thermal Anomaly or Error. Even with my old thermistor and heater evything works.
I also updated the four Prusa MK3S+ at the school and they could start printing right after.

closed 2023-03-06 by JorLars