prusa3d/Prusa-Firmware #3791: [BUG]   3.12.0-RC1-5576 PRINT FAN ERROR

wilsonayr, 2022-12-01:
**Printer type**  MK3S
**Printer firmware version** -3.12.0-RC1-5576
No modifications to hotend. Have added PrusaLink.

**SD card or USB/Octoprint**
 N/A

**Describe the bug**
  "PRINT FAN ERROR" message when running temperature calibration.
  
**To Reproduce**
Run calibration from menu.
  
**Expected behavior**
 Complete calibration
  
**G-code**
 n/a

**Video**
No video but it is a consistent error in an 18-19 deg C room.

wavexx, 2022-12-08:
You mean "PID calibration" or the new temperature model calibration?

wilsonayr, 2022-12-08:
The new temperature model calibration.

wilsonayr, 2022-12-08:
It was during the initial warm up and very early on. I don't have the serial logs at the moment but I can reload the firmware tonight to test again.

wavexx, 2022-12-08:
Interesting. If it happens very early, a video watching the extruder would be very helpful.

wavexx, 2022-12-08:
Also, please check if the fan was already spinning before the calibration is started. Normally during the heating the fan should stop immediately. I would expect the opposite to happen (right when cooldown is started the fan is started again).

@3d-gussner can you reproduce this? We have fan checks enabled during calibration, however we might have issues when intentionally triggering a fan issue

wilsonayr, 2022-12-09:
Okay I recorded a video... stopped about 3:15 in and the error occurred around 3:30. Go figure. I was able to pay better attention this time to the error. Fan was not spinning prior to starting calibration. It failed during cool down.

-----------------------------------------------------------------------------------------------------------------------------------
Here is serial log during calibration with the [echo](echo:busy: processing) lines deleted:

echo:enqueing "M310 A F0"
TM: calibration start
LCD status changed
TM: initial C est.

TM iter:0 v:14.27 e:0.248
TM iter:1 v:14.27 e:0.153
TM iter:2 v:12.92 e:0.105
TM iter:3 v:12.92 e:0.065
TM iter:4 v:12.92 e:0.040
TM iter:5 v:12.60 e:0.025
TM iter:6 v:12.60 e:0.016
TM iter:7 v:12.72 e:0.010

LCD status changed
TM: initial R 230C

TM iter:0 v:22.19 e:0.479
TM iter:1 v:22.19 e:0.296
TM iter:2 v:22.19 e:0.183
TM iter:3 v:19.68 e:0.127
TM iter:4 v:19.68 e:0.079
TM iter:5 v:19.68 e:0.049
TM iter:6 v:19.09 e:0.031
TM iter:7 v:19.09 e:0.019
TM iter:8 v:19.31 e:0.012
TM iter:9 v:19.31 e:0.007
LCD status changed
TM: cool down <50C

Print fan speed is lower than expected
LCD status changed

----------------------------------------------------------------------------------------------------------------------------------

I am unable to check the fan speed after the error occurs but the hot end fan remains running.

Leaving the printer alone after the error and the calibration resumed with "Refining estimate..." and so I let it continue at it appears to have completed.

I ran the self test and passed and decided to run the calibration one more time. This time it completed without error. I attached the full serial log from the successful calibration.

Attached
-----------------------------------------------------------------------------------------------------------------------------------------
[serial_log.txt](https://github.com/prusa3d/Prusa-Firmware/files/10192043/serial_log.txt)

wavexx, 2022-12-09:
Ok, this makes more sense. At that moment the fan is turned on at max speed, and might measure lower than expected due to the spin-up time.

Can you try to see if this test firmware:

[Firmware-MK3S-312-tm_check.zip](https://github.com/prusa3d/Prusa-Firmware/files/10194642/Firmware-MK3S-312-tm_check.zip)

does the same? It's based on the latest 3.12 branch, which is slightly newer compared to the last beta. Please try multiple times just to be sure

tjhawkey, 2022-12-10:
Thanks for posting this. I have updated a few printers recently and have seen this issue several times

wilsonayr, 2022-12-10:
Tried 3 times but was unable to flash this firmware. Had to return to RC1 each time.

--------------------------------
avrdude-slic3r -v -p atmega2560 -c wiring -P COM3 -b 115200 -D -U flash:w:0:C:\Users\Ryan Wilson\Downloads\Firmware-MK3S-312-tm_check.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Sep  6 2022 at 09:14:53
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\Ryan Wilson\Downloads\Firmware-MK3S-312-tm_check.hex"
avrdude-slic3r: writing flash (170752 bytes):

Writing | ################################################## | 100% 29.99s

avrdude-slic3r: 170752 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\Ryan Wilson\Downloads\Firmware-MK3S-312-tm_check.hex:
avrdude-slic3r: load data flash data from input file C:\Users\Ryan Wilson\Downloads\Firmware-MK3S-312-tm_check.hex:
avrdude-slic3r: input file C:\Users\Ryan Wilson\Downloads\Firmware-MK3S-312-tm_check.hex contains 170752 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 21.86s

avrdude-slic3r: verifying ...
avrdude-slic3r: 170752 bytes of flash verified

avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude-slic3r done.  Thank you.

avrdude-slic3r -v -p atmega2560 -c arduino -P COM3 -b 115200 -D -u -U flash:w:1:C:\Users\Ryan Wilson\Downloads\Firmware-MK3S-312-tm_check.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Sep  6 2022 at 09:14:53
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : arduino
                Overriding Baud Rate          : 115200
avrdude-slic3r: prusa_init_external_flash(): MK3 printer did not boot up on time or serial communication failed
avrdude-slic3r: arduino_open(): Failed to initialize MK3 external flash programming mode
avrdude-slic3r: Could not open port: COM3

avrdude-slic3r done.  Thank you.

wavexx, 2022-12-12:
@tjhawkey do you also have the same issue flashing the test FW?

tjhawkey, 2022-12-13:
I have not tried this yet (busy using my printer), but I can try tomorrow.

PR #3814 (2022-12-16): TM: reset fan measuring state when changing speed
Ensure that fan checks are reset and use the new speed at each step of the calibration.

This also gives extra time to the fan to ramp-up from a cold start, when a fancheck could previously start right *after* the speed change.

Should fix #3791

- [x] cherry-pick to MK3_3.12 branch https://github.com/prusa3d/Prusa-Firmware/pull/3825
commit c05ac535 (2022-12-16): TM: reset fan measuring state when changing speed

Ensure that fan checks are reset and use the new speed at each step of
the calibration.

This also gives extra time to the fan to ramp-up from a cold start,
when a fancheck could previously start right *after* the speed change.

Should fix #3791
PR #3825 (2022-12-16): MK3_3.12 TM: reset fan measuring state when changing speed
Ensure that fan checks are reset and use the new speed at each step of the calibration.

This also gives extra time to the fan to ramp-up from a cold start, when a fancheck could previously start right *after* the speed change.

Should fix #3791

Cherry-pick from https://github.com/prusa3d/Prusa-Firmware/pull/3814
commit 4438aa49 (2022-12-16): TM: reset fan measuring state when changing speed

Ensure that fan checks are reset and use the new speed at each step of
the calibration.

This also gives extra time to the fan to ramp-up from a cold start,
when a fancheck could previously start right *after* the speed change.

Should fix #3791