prusa3d/Prusa-Firmware #3798: Prusa Mini Firmware 4.4.0 causes preheat error with E3D's Revo Micro

FrHePo, 2022-12-04:
E3D's Revo Micro uses a 2-wire fan. The new firmware 4.4.0 doesn't take this into account. After printing the 1st layer the printer reboots and throws a preheat error. No printing is possible. 

**Printer type** - Prusa Mini
**Printer firmware version** -  4.4.0
**SD card

**To Reproduce**: Print with Firmware 4.4.0 after upgrading the extruder to E3D's Revo Micro
 
  
**Expected behavior**
  Normal printing
  
**G-code**
  This will happen with any g-code

[image]

plampix, 2022-12-05:
Which procedure did you follow to mount the revo? E3D's documentation re-uses the original fan: https://e3d-online.zendesk.com/hc/en-us/articles/4565871722397-Revo-Micro-Prusa-Mini-upgrade-guide-Edition-2-

FrHePo, 2022-12-05:
That's Edition 2 which is quite new. Edition 1 came out about last February. Edition 1 uses the 2-wire E3D-fan shipped with the Revo Micro.

3d-gussner, 2022-12-07:
@Prusa-Support @FrHePo Please move this issue to https://github.com/prusa3d/Prusa-Firmware-Buddy as this repository is for the MK2.5/S and MK3/S/+ printers