prusa3d/Prusa-Firmware #3802: [BUG]Prusa restart/Power Panic when RPI Port ON

yasars, 2022-12-05:
****Printer type** MK3S**
**Printer firmware version** -  3.11 and 3.12 beta and latest 3.12 RCI (i use it)

**Octoprint/PrusaLink**
  

****Describe the bug****
Hi if i switch my soldering iron on/off at the same socket, the prusa resets itself. but this only happens if i have the rpi port on, if i have it off and have a connection with usb, that doesn't happen.
  
****To Reproduce****
  Switch the soldering iron on/off prusa reset itself

yasars, 2022-12-06:
Hi the raspberry os printed over rpt port the its restart ,.over usb no problem... The problem is when i will usw rpi port... I have check my.connection all fine...

Prusa-Support, 2022-12-28:
Thanks for reaching us.

Maybe the RPi port draws some power, draining power from PSU/board more quickly, and making the printer more sensitive to small power oscillations/drops.
However, I'm still not sure if the Power Panic is triggered, that would be noticeable after the reboot.
The problem though may seem related to the socket or the power strip.

Are the devices connected together on the same power strip?
Would the problem persist in the same way on a different power strip, or if connected to a different wall socket?

If the Power Panic is triggered, you will hear a beep after the reboot, and an option to recover the print will show up.
Would a Power Panic recovery message actually show up after the printer reboot? 

It could be useful to be able to reproduce the issue with other power strips or devices else than that soldering iron.

Michele Moramarco
Prusa Research

yasars, 2022-12-28:
Hi Michele , thank you ! 

i have observed the problem, i only have a soldering iron and a hot air dryer and it really only happens on this socket strip and only with the soldering iron.... it's not a power panic... i looked at the counter under > support it always stays stand at 0 and it doesn't beep either....

 thank you very much ...

yasars, 2023-02-05:
And now.after update to the  new stable ... The same error is comming  with usb too....

Prusa-Support, 2023-02-16:
We are aware that when the printer is connected via USB or RPi port, the Power Panic doesn't work at the moment. Probably that's why the Power Panic recovery is not triggered.
Maybe a Power Panic recovery will be triggered while you print on the standalone printer, via LCD panel, while RPi port is OFF and any third-party device (USB or interface pins) is disconnected.

The problem seems to be related to your specific setup though.
You can try reading the output voltage on a power strip outlet while you turn on/off the soldering iron to see if the voltage is altered.
There is a chance that it is causing interference to a portion of the mains grid, connected to the same electrical panel. You can try to read voltage on some nearby wall sockets, while someone switches the soldering iron ON/OFF.

Michele Moramarco
Prusa Research

yasars, 2023-02-16:
the prusa runs on its own power connection (circuit breaker)... i'll probably have to unscrew the soldering iron and have a look... or i'll exchange it even though it's almost new... thank you

Prusa-Support, 2023-02-24:
Maybe the soldering iron functionality is interfering with a relatively large portion of the grid.

It is still not 100% unclear why it affects the printer only if RPi port is ON but I think it simply draws more power.
The printer may not be the only device subject to the soldering iron interferences though.

Please keep us posted if you don't mind.
We haven't got similar reports so far.

Michele Moramarco
Prusa Research

moto52000, 2023-03-07:
Hello,  I have a very similar problem.  RPi is connected to Prusa MK3S+ with the supplied USB cable running Octoprint.  Often (not every time) when I remove the build plate the PI shuts down. Other things have caused this as well (like just walking by the printer) and it seems to be static interference.  Maybe?  
I have changed the power supply and am now using the supply that comes with the PI.  
Moved the printer to another room
The only thing still on is the soft shutdown button and LED which I will remove next.  
Any ideas?

Prusa-Support, 2023-03-28:
We may be looking at a different problem.
The cause of electrical interference described in the main post is caused by other devices sharing the power grid with the printer: it is an external problem on the grid.
The reason why the problem only occurs when RPi port is enabled is not totally clear though. However, It is a fact that a device connected via USB or RPi port will draw power from the printer and may empty the printer's capacitors too fast. Also, we know that the Power Panic feature may not work correctly as a consequence.

In the last reported problem, the electrical interference may not become from the grid, but may be caused by:
a) additional third-party devices connected to the printer, not specifically designed or approved to work with the printer;
b) how the printer is connected to the grid (static charges and shocks because the printer is not correctly grounded).
Disconnecting the additional third-party devices (maybe one for a time) is surely a good troubleshooting practice to track back a connected device that may be the root cause of the problem.
Also please make sure the printer is correctly grounded. Some wall sockets or plug adapters may not ground the printer correctly or may be missing a ground channel.

Michele Moramarco
Prusa Research