prusa3d/Prusa-Firmware #3809: [BUG] MK3S 3.11.0 compiled firmware differ from Prusa release

gregsaun, 2022-12-06:
**Printer type** - MK3S
**Printer firmware version** -  3.11.0
**MMU upgrade** - No

**Describe the bug**
When I compile the firmware using PF-build.sh I am unable to match the original Prusa firmware binary available on github or prusa3d.com. The file size and checksum are different. It looks to be the same problem as reported in https://github.com/prusa3d/Prusa-Firmware/issues/2799 but for 3.11.0. I have tried several PF-Builds options without success. I am therefore unable to verify that my build doesn't have problem nor to understand what are the differences.

File size is different:
```
-rwxrwxrwx 1 greg greg 1056934 Dec  6 20:32 prusa_firmware/PF-build-hex/FW3110-Build4955/BOARD_EINSY_1_0a/FW3110-Build4955-1_75mm_MK3S-EINSy10a-E3Dv6full.hex
-rwxrwxrwx 1 greg greg 1056965 Apr  7  2022 prusa_firmware/fw_original/prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex
```

md5sum checksum:
```
5cfcd604d3c15e5b974f91172dbb5165 prusa_firmware/fw_original/prusa3d_fw_MK3S_MK3S+_3_11_0_4955.hex
5f59f2430f3fbf719cf78a5a7d5c9d07 prusa_firmware/PF-build-hex/FW3110-Build4955/BOARD_EINSY_1_0a/FW3110-Build4955-1_75mm_MK3S-EINSy10a-E3Dv6full.hex
```

diff from a hexdump:
```
original firmware                                                                     | built firmware
000140e0  34 36 34 31 34 45 30 30  34 31 37 30 37 32 32 30  |46414E0041707220|        | 000140e0  34 36 34 31 34 45 30 30  34 34 36 35 36 33 32 30  |46414E0044656320|
000140f0  32 30 33 37 32 30 42 44  0d 0a 3a 31 30 37 32 32  |203720BD..:10722|        | 000140f0  32 30 33 36 32 30 44 35  0d 0a 3a 31 30 37 32 32  |203620D5..:10722|
---
00014160  45 36 36 36 39 36 37 32  39 30 30 34 31 37 30 37  |E666967290041707|        | 00014160  45 36 36 36 39 36 37 32  39 30 30 34 34 36 35 36  |E666967290044656|
00014170  32 32 30 32 30 33 37 32  30 33 32 33 30 45 36 0d  |2202037203230E6.|        | 00014170  33 32 30 32 30 33 36 32  30 33 32 33 30 46 45 0d  |3202036203230FE.|
---
00014190  33 31 33 31 33 41 33 30  33 39 33 41 33 35 33 31  |31313A30393A3531|        | 00014190  33 32 33 30 33 41 33 32  33 37 33 41 33 34 33 31  |32303A32373A3431|
000141a0  30 30 32 30 33 33 32 45  33 31 35 33 0d 0a 3a 31  |0020332E3153..:1|        | 000141a0  30 30 32 30 33 33 32 45  33 31 35 34 0d 0a 3a 31  |0020332E3154..:1|
---
0001a710  30 34 31 37 30 37 32 46  31 0d 0a 3a 31 30 39 36  |0417072F1..:1096|        | 0001a710  30 34 34 36 35 36 33 30  38 0d 0a 3a 31 30 39 36  |044656308..:1096|
0001a720  37 30 30 30 32 30 32 30  33 37 32 30 33 32 33 30  |7000202037203230|        | 0001a720  37 30 30 30 32 30 32 30  33 36 32 30 33 32 33 30  |7000202036203230|
---
0001a740  36 36 37 35 34 41 0d 0a  3a 31 30 39 36 38 30 30  |66754A..:1096800|        | 0001a740  36 36 37 35 34 42 0d 0a  3a 31 30 39 36 38 30 30  |66754B..:1096800|
---
001020a0  30 31 46 46 0d 0a 31 5f  37 35 6d 6d 5f 4d 4b 33  |01FF..1_75mm_MK3|        | 001020a0  30 31 46 46 0d 0a                                 |01FF..|
001020b0  53 2d 45 49 4e 53 79 31  30 61 2d 45 33 44 76 36  |S-EINSy10a-E3Dv6|        | 001020a6
001020c0  66 75 6c 6c 0a                                    |full.|                   <
001020c5                                                                              <
```

It seems that adding the `1_75mm_MK3S-EINSy10a-E3Dv6full` with the `CR` ASCII character is fixing the file size but there are still differences in the binary. I couldn't find a PF-build.sh option that have added this at the end of the hex file.
  
**To Reproduce**
To test all possibilities and be sure I was not adding human errors I made a bash script to build and compare the files. I am running this script using WSL with an Ubuntu distrib.

The bash script (remove the .txt extension):
[build_and_compare.sh.txt](https://github.com/prusa3d/Prusa-Firmware/files/10170160/build_and_compare.sh.txt)

WSL version:
```
> wsl -l -v
  NAME      STATE           VERSION
* Ubuntu    Running         2
```

Ubuntu version:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.1 LTS
Release:        22.04
Codename:       jammy
```

**Expected behavior**
The hex file built should have same size and checksum than the original Prusa hex file.

**The end**
Thank you for your help :)

3d-gussner, 2022-12-07:
@gregsaun Yes you are right you will never get identical hex files with the current build platforms we are using. Even if you build same code twice locally you will see a difference, which is the time of compiling.
Arduino IDE on Linux and Windows compiles differently, paths are another factor which changes the output.

This is one of the reasons we started to add `cmake` build, which hopefully will be the main build environment with FW3.13.0 and newer.

To get the same hex file as Prusa provides you would need to use the same OS, scripts, username, paths and modify time to the compilation time of the official build.
To be honest  I gave up on this and happy that the `cmake` build will improve things.

I guess you want to update your custom firmware for the BEAR upgrades, the `PF-build.sh` and `build.sh` builds are perfectly fine.

All build platforms use same
- Arduino IDE version
- Arduino_Boards
- settings

Please look at the `cmake` builds for the future releases.

As I wrote the `PF-build.sh` script to make compiling easier, I will try to support it for two or three releases to give the modding community time to adjust to `cmake` builds.

3d-gussner, 2022-12-07:
Yes these are correct, but I would advise to create copies of the variants and change there the different settings so your changes aren't overwritten next time you sync with the Prusa repository. Do you have a fork?

The Board mem and flash messages should be fixed, which release do you try to build?

BTW: These are special for the MK404 sim to allow ATmega404 with lots of RAM and FLASH for debugging.

Side note the MK404 has a `-g bear` version :wink: 
[image]

gregsaun, 2022-12-07:
Thanks! Yes sure we are using a fork and automated build process ;) : https://github.com/bear-lab-3d/Prusa-Firmware/ . We use Drone CI, very close to github actions, you can see the build process here: https://github.com/bear-lab-3d/Prusa-Firmware/blob/BEAR_MK3S/.drone.yml .

Regarding the board mem/flash message I am using 3.11.0 at the moment so if I understood correctly those message are still present but will be fixed in later version. Could you confirm that answering *yes* for both is correct?

Yes I am aware of the MK404 with Bear, VintagePC has integrated it, very nice work. I haven't tried yet on WSL2 (I have just finished to build up a new computer).

3d-gussner, 2022-12-07:
It might be that the FW 3.11.0 PF-build.sh has the fix for the board mem/flash. Newer firmware version should be fixed with https://github.com/prusa3d/Prusa-Firmware/pull/3422. FW release 3.12.0-BETA1 and newer shouldn't have this issue anymore.

The PF-build.sh calls the MK404-build.sh at the end.

gregsaun, 2022-12-07:
As far as I understand, 3.11.0 has the messages as I get them when I use this:
```
git reset --hard HEAD
git clean -f -d
git checkout MK3_3.11.0
./PF-build.sh -v 1_75mm_MK3S-EINSy10a-E3Dv6full.h -l ALL -d GOLD
```

Terminal output:
```
...
Printer        : MK3S                                                                                                                                                                                                                         
Variant        : 1_75mm_MK3S-EINSy10a-E3Dv6full                                                                                                                                                                                               
Firmware       : 3110                                                                                                                                                                                                                         
Build #        : 4955                                                                                                                                                                                                                         
Dev Check      :                                                                                                                                                                                                                              
DEV Status     : GOLD                                                                                                                                                                                                                         
Motherboard    : BOARD_EINSY_1_0a                                                                                                                                                                                                             
Board flash    :                                                                                                                                                                                                                              
Board mem      :                                                                                                                                                                                                                              
Languages      : ALL                                                                                                                                                                                                                          
Hex-file Folder: PF-build-hex/FW3110-Build4955/BOARD_EINSY_1_0a                                                                                                                                                                               
Hex filename   : FW3110-Build4955-1_75mm_MK3S-EINSy10a-E3Dv6full                                                                                                                                                                              
                                                                                                                                                                                                                                              
 
Multi-language firmware will be built
 
Board mem has been already modified or not reset
Current: 0x21FF
1) Yes
2) No
Select Yes if you want to reset it.1
Resetting board mem size
Board flash has been already modified or not reset
Current flash size: 0x3FFFF
Current max.  size: 253952
1) Yes
2) No
Select Yes if you want to reset it.1
Resetting board flash size

Start to build Prusa Firmware ...
Using variant 1_75mm_MK3S-EINSy10a-E3Dv6full
...
```

3d-gussner, 2022-12-07:
Yes FW 3.11.0 hasn't that fix. BTW please try to reset or checkout to the tags as branches may change while a tag don't.

```
git remote -v // ensure that you have https://github.com/prusa3d/Prusa-Firmware.git as upstream
git fetch --all // to get all updates
git reset --hard v3.11.0 // or git checkout v3.11.0
```

gregsaun, 2022-12-07:
I have updated thank you and no changes so far to the binary but it is normal as we discussed here.

Thanks for your help!

closed 2022-12-07 by gregsaun