prusa3d/Prusa-Firmware #3815: [FEATURE REQUEST] beep when preheating has reached desired temperatures

roblatour, 2022-12-10:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]

**Is your feature request related to a problem? Please describe.**
  A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
  
**Describe the solution you'd like**
  A clear and concise description of what you want to happen.

On the printer's control panel there is a feature to preheat (the bed and nozzle).  It would be nice if when the preheating has reached the temperatures required that the printer beeps (much like it does when it is ready for you to unload the filament).

Prusa-Support, 2022-12-29:
Thank you for sharing your idea.

Our developers will think about it.

For now, please consider that the nozzle may take ≈2 minutes to reach temperatures ≈270°C or so, and the bed may take ≈6 minutes to reach ≈100°C. The time may vary for several reasons.
You may want to check back the printer after ≈90 seconds if you care about the nozzle temperature, or ≈5 minutes if you care about the bed temperature.

Michele Moramarco
Prusa Research

roblatour, 2022-12-29:
Hi Thanks.

Regarding >>

For now, please consider that the nozzle may take ≈2 minutes to reach temperatures ≈270°C or so, and the bed may take ≈6 minutes to reach ≈100°C. The time may vary for several reasons.
You may want to check back the printer after ≈90 seconds if you care about the nozzle temperature, or ≈5 minutes if you care about the bed temperature.
That is exactly why I am making the suggestion.  In that time it takes the nozzel and bed to get up to temperature I would like to be doing other things without needing to check my printer, and adding the beep would let me know it is done without me having to keep checking it.

Thanks for your consideration.

Rob

Prusa-Support, 2023-01-09:
Hello Rob, thank you for the request. Passed it on to our devs for them to evaluate - it might unfortunately have low priority at this time, we can't promise we'll implement it, but it will surely be evaluated. 🙂 

Alessandro Pantaleo
Prusa Research

cnlson, 2023-01-11:
uh, why not add a buzzer sound to the start gcode? 

this is in my ending gcode to beep when a print is done
M300 S294 P220 ;N1 D4 1/8
M300 S20000 P20
M300 S294 P220 ;N1 D4 1/8
M300 S20000 P20
M300 S294 P220 ;N1 D4 1/8
M300 S20000 P20
try it inbetween m109 and g28. I am fairly sure until it hits temp, it pauses. i have mine heat E to 140 and then fully heat the bed then finishing heating the extruder (so it isn't spending 5 minutes dripping filament) 

M104 S140
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level

so this would become 
```
M104 S140
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
M300 S294 P220 ;N1 D4 1/8
M300 S20000 P20
G28 W ; home all without mesh bed level
```

roblatour, 2023-01-12:
I'm not sure you understand.  I want to press the button on the Prusa printer to start the pre-heat process, and when it is done have the printer beep.   

Often times I am working on other things, like still fiddling with the print in slicer, adding supports etc., while the printer is warming up.  The idea is that I want to know, without having to keep looking at the screen on the printer, when the pre-heat is done.  That is why the beep would be of help.

cnlson, 2023-01-12:
I get what you want, you could certainly make a gcode just of the starting
sequence so that starting it just does a preheat and beeps. At least you'll
have the functionality you want. They aren't going to implement this for a
long time if ever in the firmware.

roblatour, 2023-01-12:
"They aren't going to implement this for a long time if ever in the firmware." - you just need faith :-)_

cnlson, 2023-01-12:
I've been waiting for tree supports for years.

If I could have had it with a gcode file immediately I wouldn't have waited

Prusa-Support, 2023-01-21:
The use of specific gcodes for the individuals' specific needs is surely a legit alternative. Setting a 5-6 min timer via phone assistant could be efficient and quick enough too. Other than that, this issue will remain open as inspiration for future developments.

As a company, we need to set priority on tasks (and requests) in accordance with the company goals, nonetheless, we are most attentive to the needs of our large community too.
A big and most challenging task of our developers is making room in the Einsy memory for new features too (without breaking anything).
Please bear with us and thanks in advance for your understanding.
 . 
 
By the way, not only "tree" but "organic supports" are on their way.
https://www.youtube.com/watch?v=ThiuzonYH10&t=690s
This is definitely off-topic though.

Michele Moramarco
Prusa Research

closed 2023-11-13 by github-actions[bot]