prusa3d/Prusa-Firmware #3820: M852 Reportback Unknown M code

Ethan-Ming, 2022-12-13:
i found my print is still skewed depside xyz calibration detects it 

so i was trying us correct it by M852 but got this  

Send: M852 I-0.000334
Recv: Unknown M code: M852 I-0.000334
Recv: ok

wavexx, 2022-12-13:
> i found my print is still skewed depside xyz calibration detects it

AFAIK if the skew is small the correction is intentionally skipped. Does the reported skew matches your measurement, or you're just manually measuring?

M852 is not supported. But it looks like it wouldn't be hard to support ``M852 [I] [S]`` since the skew matrix is already there.

Ethan-Ming, 2022-12-15:
yes the reported skew matches my measurement

printer reported 0.1° skew on LCD and is the same as what I measured on the physical print part using Micrometer

Ethan-Ming, 2022-12-16:
> AFAIK if the skew is small the correction is intentionally skipped. Does the reported skew matches your measurement, or you're just manually measuring?
>
> M852 is not supported. But it looks like it [...]

Prusa-Support, 2023-01-10:
Thanks for your suggestion.
Our developers will think about it.

For now, you should be able to improve the first layer's accuracy by enabling the _7x7 mesh grid_ via [Mesh Bed Levelling settings](https://help.prusa3d.com/article/mesh-bed-leveling_112163) .
In some cases, the [Bed Level Correction](https://help.prusa3d.com/article/bed-level-correction_2267) can also help make it up for minor additional inconsistencies across the edges.

Michele Moramarco
Prusa Research

wavexx, 2023-01-10:
Note that XY skew can't really be corrected or improved via MBL.

AFAIK the choice of not enabling skew correction for small skew angles was/is intentional looking at the code (I guess there is a limit in the how accurately we can detect skew - but I'm not sure this is the only reason).

leptun, 2023-01-10:
> AFAIK the choice of not enabling skew correction for small skew angles was/is intentional looking at the code

It was done like this most likely because it can create artifacts during the print. On a mostly straight printer, a long extrusion along the Y axis might create visible steps in the surface because at low skew angles, the distance between each microstep in the X axis adjustment is quite large. I may have gotten the X and Y axis in reverse, but the principle is the same.

Prusa-Support, 2023-01-21:
I guess this issue can stay open as a possible feature request to support `M852` code. I'm not sure if these kinds of corrections are necessary though.

Not sure if we are looking at the calibration numbers for analysis purposes, or at actual first layer quality issues. In the latter case, our official Tech Support would be able to help, and you could consider closing the issue.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

closed 2023-11-13 by github-actions[bot]