prusa3d/Prusa-Firmware #3856: [BUG] Buzzer sound M300 S0 broken

sarusani, 2022-12-28:
**Printer type** - MK3S
**Printer firmware version** -  3.12.0-RC1

**SD card or USB/Octoprint**
  Both

**Describe the bug**
  Since 3.12.0-RC1 my "Print has finished" buzzer sound does not sound correct anymore. I think it has something to do with a different handling of M300 S0. It used to be a pause, now it generates a tone. (Other Marlin printers handle it as a pause as well)

**To Reproduce**
  Run the following gcode: (terminal or file)
  ```
  ;Indiana Jones
  M300 S330 P226
  M300 S0 P120
  M300 S349 P115
  M300 S392 P111
  M300 S0 P120
  M300 S52 P1034
  M300 S0 P120
  M300 S294 P226
  M300 S0 P120
  M300 S330 P115
  M300 S34 P1264
  M300 S0 P120
  M300 S392 P226
  M300 S0 P120
  M300 S440 P115
  M300 S494 P111
  M300 S0 P120
  M300 S69 P1034
  M300 S0 P120
  M300 S440 P226
  M300 S0 P120
  M300 S494 P115
  M300 S523 P462
  M300 S587 P462
  M300 S659 P226
  M300 S0 P236
  ```
  
**Expected behavior**
  Same output as before installing 3.12.0-RC1 or a way to make M300 S0 work as a pause again.

leptun, 2022-12-28:
M300 was changed so that if the frequency is not provided or 0, the printer will not modulate the buzzer, but instead use its native frequency. The sound produced by M300 S0 is considerably crisper than anything that can be generated by other frequencies. Previously, there was no way to generate a beep with no modulation, hence why S0 was chosen as an exception.
If you want to create delays, see the G4 command.

wavexx, 2022-12-28:
We could have different behavior for omitted and S0. We could make S0 explicitly pause, and have [omitted] work as it is now. Not sure if that would be a good idea or not.

Marlin uses an explicit default: https://marlinfw.org/docs/gcode/M300.html

there is no mention of the behavior when S0 is used.

sarusani, 2022-12-28:
Thanks for the info!
I replaced the M300 S0 with G4, which works fine. (ex. M300 S0 P120 turns into G4 P120)

Assuming the firmware is based on Marlin 1.0.x this should behave exactly the same as S0 did. (The non blocking tone queue was introduced in 1.1.x)

sarusani, 2022-12-30:
> We could have different behavior for omitted and S0. We could make S0 explicitly pause, and have [omitted] work as it is now. Not sure if that would be a good idea or not.
>
> Marlin uses an explicit [...]

Just to give my 2 cents... :)
I'd prefer to keep the "standard" behavior of S0 and would only trigger high when S is omitted. Just to keep it in line with other Marlin based firmware.
But after all the tone feature is just a "nice to have" and I don't think anybody will be mad if you leave it the way it is implemented right now.

Feel free to close this issue.

Thanks again!

wavexx, 2022-12-30:
Diverging with marlin is always bad so let's try to follow it when possible even when underspecified

wavexx, 2022-12-30:
@sarusani can you test PR #3869?
I tried to play the indy theme. Wow some frequencies are horrible ;), but it does pause correctly

commit e45adb4c (2023-01-02): Make M300 S0 pause like Marlin

Follow Marlin's behavior and simply insert a delay for the requested
duration when using M300 S0.

When S is not specified, use the default tone instead.

Fixes #3856
commit 805e71e5 (2023-01-02): Make M300 S0 pause like Marlin

Follow Marlin's behavior and simply insert a delay for the requested
duration when using M300 S0.

When S is not specified, use the default tone instead.

Fixes #3856
closed 2023-01-02 by wavexx