prusa3d/Prusa-Firmware #3860: [BUG]<User inputs not recognized during crash detection>

Sarlochin, 2022-12-29:
**Printer type** - [MK3S]
**Printer firmware version** -  [3.11.0-4955]

**MMU upgrade** - [none]
**MMU upgrade firmware version** - [n/a]

**SD card or USB/Octoprint**
 -> PrusaLink 0.7.0rc2

**Describe the bug**
 When a crash is detected during printing, it is not possible to pause the print to resolve the issue. The pause print entry can be selected, but the printer will re home and crash until the print fails and pauses automatically. It is therefore not possible to prevent repeated crashes.
  
**To Reproduce**
  Crash during printing and select the "pause print" menu entry
  
**Expected behavior**
Print can be paused by the user if the crash is first detected.
The printer detects a crash and re homes. During that process, the “pause print” menu entry is selected by the user.
The printer goes into parking position until user intervention is concluded. 
Thereafter, the print can be resumed.
  
**G-code**
 not G-code specific

**Video**
n/a

Sarlochin, 2022-12-29:
Related to #607 prusa3d/[Prusa-Link]

Prusa-Support, 2023-01-11:
Thank you for your suggestion.

In many cases, the printer will safely restore from an accidental crash so the automatic pause after three (I believe) crashes is an intentional feature.
The inability of the user to correctly and timely pause a print to solve an issue right away though, it's surely a disadvantage.

This may not be a bug, but a request to improve the current printed behavior in this scenario. I'll add the relevant label and leave this up to the developers.

Thank you again, our developers will look into this.

Michele Moramarco
Prusa Research

closed 2024-01-22 by github-actions[bot]