prusa3d/Prusa-Firmware #3870: [FEATURE REQUEST] During crash detection/power panic, raise Z-height above highest previously printed z-height to avoid repeated crashes when "complete individual objects" is on

danchar4, 2022-12-31:
**Printers affected**

- MK3s+
- MK2.5S
- Mini+

**Is your feature request related to a problem? Please describe.**

I have a plate full of identical parts of 80mm height (higher than the X-axis rod) printing with the "sequential printing - complete individual objects" in PrusaSlicer. Extruder clearances are set with a larger margin than needed. If all goes well, the print succeeds and the parts look great.

But if there is a power panic or crash detected after the first part completes, the extruder will collide into the first printed part over and over again eventually warping it rendering the whole run ruined. There is nothing special about the parts being printed - they can be solid cubes or whatever, so I haven't attached .stl or .3mf since it wouldn't be helpful.

[image]

**Describe the solution you'd like**

What I expect is the following:

- Start print in sequential mode (identical parts front-to-back, left-to-right per [prusa3d.com guidance](https://help.prusa3d.com/article/sequential-printing_124589))
- First part succeeds
- During second part, a crash or power panic is detected
   - printer stores current Z-height and highest previously printed Z-height
- Recovery
   - raise Z **above highest previously printed Z-height or to height specified in PrusaSlicer** and relayed to printer via g-code metadata or startup code
   - home X and Y
   - if Z-height cannot be guaranteed after a crash, re-calibrate Z
      - raise Z to top to equalize Z motor heights
      - move X/Y to rear-right where (hopefully) nothing has been printed yet and no risk of collision
      - home Z to bed
   - raise Z back up
- Resume print

**Additional thoughts**
I suspect the "fix" is in firmware where the printer needs to know that it is in sequential mode and the max Z-height of the print via metadata stored in G-code - then it can run an augmented recovery routine. But maybe I'm just using PrusaSlicer wrong? Or maybe I have unrealistic expectations of crash recovery/power-panic capabilities? 

**Related Discussion**
https://www.reddit.com/r/prusa3d/comments/zxn363/mk3s_complete_individual_objects_crash/

wavexx, 2022-12-31:
See #3419

danchar4, 2022-12-31:
Closing as duplicate of #3419 
Thanks @wavexx

closed 2022-12-31 by danchar4