prusa3d/Prusa-Firmware #3872: [Bug] 3.12.0-RC1 filament sensor forcing unload/reload during prints

cristian-rivera, 2022-12-31:
Hi, I have a brand new MK3S+ kit that is about a month old. It runs perfectly on 3.11.0 with no issues. The v4 filament sensor is properly detected using Settings > HW Setup... and the voltages are either 0.3 or ~4 when either loaded or unloaded.

I decided to flash 3.12.0-RC1 because I was interested in the new PID calibration (which works great btw) and I made sure to use the correct version for the MK3S+.

During every print, after a somewhat random amount of time, the print will halt and the machine will force me to unload and reload the filament. If the print is long enough, this happens multiple times a print. Filament autoloading works just fine when a print is not in progress.

I was reading the release notes for 3.12.0-RC1 and it states:

> A feature detecting the filament sensor hardware version is currently not working. Printing process isn't affected, however if the filament sensor is not working properly, the printer is unable to verify it. The devs are aware of this issue and a fix will be provided soon.

This part specifically:

> however if the filament sensor is not working properly, the printer is unable to verify it

led me to beleive that if everything on 3.11.0 IS woring perfectly fine, then things will work just fine with the filament sensor in 3.12.0-RC1, but that doesn't seem to be the case.

Also:

> Printing process isn't affected

unfortunetely for me, the printing process is affected. Is this a known issue internally? I tried looking through open issues on the repo, but I can't find anything seemingly related to the issue I'm facing.

For now, I've reflashed 3.11.0 and things are back to normal with prints not halting anymore!

wavexx, 2023-01-02:
Can you be more precise in how frequently does it happen? Minutes? Hours?

cristian-rivera, 2023-01-04:
Hi @wavexx, I've just flashed 3.12.0-RC2 and I'm going to run a few prints to see if the problem still occurs. If it does, I'll report back the frequency!

cristian-rivera, 2023-01-05:
I think we can close this. RC2 fixed it as far as I could tell. I've been running the printer off and on for about 24 hours and the issue has not come back. I looked at the RC1 -> RC2 diff and I couldn't find something obvious that could've fixed this, but I'm confident it's gone now.

cristian-rivera, 2023-01-05:
I'm sorry @wavexx , I guess I jinxed myself. I've run multiple 1-4 hour prints over the past 24 hours with no issues and then all of the sudden, my current 4 hour print promted a filament change with only ~15 minutes left. Is there any debug data I could retrieve for you?

ndng28, 2023-01-13:
I have updated to the latest FW and this behavior has happened twice during the last two prints.

cristian-rivera, 2023-01-13:
I’m glad I’m not the only one who has this issue. I’m very surprised that 3.12.0 was allowed to be released with this known issue in the release notes since 3.12.0-RC1.

> Known issues
> 
> A feature detecting the filament sensor hardware version is currently not working. Printing process isn't affected, however if the filament sensor is not working properly, the printer is unable to verify it. The devs are aware of this issue and a fix will be provided soon.

Especially since this is incorrect as it has failed many of my prints with a filament sensor that worked perfectly fine on 3.11.0.

> Printing process isn't affected

DRracer, 2023-01-13:
I'm sorry to hear that your issue is back. Unfortunately, we have been unable to reproduce your issue so far, but we got another report like your today.
The temporal removal of filament sensor board version detection code should not influence the reliablity of filament sensor itself. The detection algorithm will return in 3.13 shortly.

Please stay tuned, we'll do some more investigation.

ndng28, 2023-01-13:
I am using USB/Octoprint

Prusa-Support, 2023-02-02:
Thanks for reporting as you already know, the filament sensor detection is temporarily broken.
Please notice though: `filament sensor detection` ≠ `filament detection`.
The printer can no longer debug the filament sensor, but a proper troubleshooting session will.

There are good chances that your filament sensor works correctly but debris interferes with the triggering mechanism or a loosened cable causes signal interruptions. I've also observed similar issues occurring due to hardly reproducible assembly inconsistencies (deformed or misaligned parts, incorrectly tightened bolts...). Even a too-thin or deformed (ground) filament may trigger a filament outage. Also please check if `M600` is unintentionally added in the gcode.

You can downgrade to 3.11.0 and use the filament sensor detection to debug the filament sensor.
The filament sensor recognition is explained in [FW 3.9.0 release notes](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.0).
At the same time, you can run several prints with FW 3.11.0 and see if the fake filament outage is reproducible.

Most likely, debris is interfering with the triggering mechanism. If the problem will persist with 3.11.0, please consider reaching the official Tech Support for a proper troubleshooting session.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

cristian-rivera, 2023-02-02:
> Thanks for reporting as you already know, the filament sensor detection is temporarily broken. Please notice though: `filament sensor detection` ≠ `filament detection`. The printer can no longer [...]

Thanks for the reply. I can assure you that this is not an issue with debris. I have dissasembled and reassembled the entire hotend to clean the sensor and make sure all related parts are working properly. I have also upgraded to and downgraded from 3.12.0 many times, while also factory resetting the mainboard and the issue is only happening on 3.12.0, not 3.11.0. The filament sensor itself works just fine, the issue here is something related to the removal of the sensor detection in 3.12.0.

dlozinski, 2023-02-04:
This bug is still present on firmware **3.12.1**.
When I flashed 3.11.0 everything goes back to normal, however I noticed short 3 short (1 sec) pauses during the print, that may be caused be PI Zero slow CPU. Firmware 3.11 ignores that, but 3.12 probably detects filament not moving?

I have Prusa MK3 and I print from octopi.

MikeEckels, 2023-02-12:
Is this being worked on? This is a major issue for me as well. My printer constantly tries to change filament mid print. Never happened before my upgrade to 3.12.0

cristian-rivera, 2023-02-12:
I’m glad more people are coming forward with this same issue. I felt that in the beginning I wasn’t being taken seriously, but there is certainly something going on here and I’m positive it has nothing to do with debris.

I am also extremely shocked 3.12.0 was released with “filament sensor detection is broken”. I get it’s “supposed” to not cause side effects, but releasing something mainstream with a quite important feature known to be “broken” is quite eye opening, especially from Prusa.

ndng28, 2023-02-12:
I have cleaned the IR sensor for "debris" and the issue persists. It was resolved when I downgraded it to the last version. 

Pretty frustrated here as well...I had to turn off crash detection to resolve a layer shift issue with ABS and now filament sensor too.

Prusa-Support, 2023-02-15:
Thanks for confirming that the issue won't persist after downgrading to FW 3.11.0. That also confirms this is firmware related.

As of now, we know that the filament sensor detection is broken, and we will work on that.
As far as we know, that won't cause the filament sensor to malfunction during everyday printing, but we may be missing something.

We will fix the detection and investigate the filament sensor anomalies meanwhile.

Michele Moramarco
Prusa Research

PyBonnetainNesterenko, 2023-04-06:
Could this be the same issue as in https://github.com/prusa3d/Prusa-Firmware/issues/4061 ?

W1SPS, 2023-04-10:
I have the same issue. on my MK3s+.  I wake up to the printer beeping asking me to unload filament, then asks me to re-insert filament, then asks if it loaded OK (I hit yes), then it resumes the print. If it's a long enough print It may (or may not) do this several more times.  Just Started doing this in the last week. I use Octoprint, and all software is latest release.

Prusa-Support, 2023-04-22:
We have seen this problem with any firmware version before now.
The cause is most often debris, assembly, or connection related.
There could be other reasons too and https://github.com/prusa3d/Prusa-Firmware/issues/3872#issuecomment-1413930200 covers a few popular cases.
Our Customer Support would be able to help you properly test the filament sensor functionality.
https://help.prusa3d.com/article/customer-support_2287

We can't reproduce the issue but if you alternate FW 3.12.2 and FW 3.11.0 from time to time, and make _absolutely_ sure you can still reproduce the issue on **FW 3.12.2 only** - 👉  _0% occurrences on FW 3.11.0_ 👈 - we can assume we are looking at a bug.

Please keep testing and reporting, but also consider contacting our Customer Support.

Michele Moramarco
Prusa Research

cristian-rivera, 2023-04-22:
> We have seen this problem with any firmware version before now. The cause is most often debris, assembly, or connection related. There could be other reasons too and [#3872 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/3872#issuecomment-1413930200) covers a few popular cases. Our Customer Support would be able to help you properly test the filament sensor functionality. https://help.prusa3d.com/article/customer-support_2287
> 
> We can't reproduce the issue but if you alternate FW 3.12.2 and FW 3.11.0 from time to time, and make _absolutely_ sure you can still reproduce the issue on **FW 3.12.2 only** - 👉 _0% occurrences on FW 3.11.0_ 👈 - we can assume we are looking at a bug.
> 
> Please keep testing and reporting, but also consider contacting our Customer Support.
> 
> Michele Moramarco Prusa Research

Hi Michele, I can assure you this is not related to debris, assembly, or connection. I am absolutely clear that this issue is not on 3.11.0. I don't know how I could be more clear. Others have reported the exact same issue. The odds that we see the problem only on 3.12, but it actually being due to all of our printers having debris, assembly or connection issues is impossible.

MikeEckels, 2023-04-22:
> > We have seen this problem with any firmware version before now. The cause is most often debris, assembly, or connection related. There could be other reasons too and [#3872 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/3872#issuecomment-1413930200) covers a few popular cases. Our Customer Support would be able to help you properly test the filament sensor functionality. https://help.prusa3d.com/article/customer-support_2287
> > We can't reproduce the issue but if you alternate FW 3.12.2 and FW 3.11.0 from time to time, and make _absolutely_ sure you can still reproduce the issue on **FW 3.12.2 only** - 👉 _0% occurrences on FW 3.11.0_ 👈 - we can assume we are looking at a bug.
> > Please keep testing and reporting, but also consider contacting our Customer Support.
> > Michele Moramarco Prusa Research
> 
> I would like to confidently assert that the issue we are experiencing is not related to debris, assembly, or connection. After thorough investigation, I am certain that this issue does not pertain to version 3.11.0. Multiple users have reported encountering the same problem, which further supports this conclusion.
> 
> Considering the prevalence of the issue exclusively in version 3.12, it is highly improbable that debris, assembly, or connection issues are the root cause for all our printers. I kindly request that you reconsider this possibility and join us in finding a more plausible explanation for the problem.

Absolutely agree, I have been running 3.11.0 for over 2 months now, and have not had ANY issues with this. As soon as I switch back to the newest version (3.12.X) I receive these errors within hours of operation.

@Prusa-Support Please understand that this is NOT a hardware issue. We have far to many users reporting the exact same. This is beyond a doubt software related.

PyBonnetainNesterenko, 2023-04-27:
> Absolutely agree, I have been running 3.11.0 for over 2 months now, and have not had ANY issues with this. As soon as I switch back to the newest version (3.12.X) I receive these errors within hours [...]

Same stuff here. Running 3.11.0 : absolutely no problem. Installing 3.12.X, « filament out » wrongly triggered several times per print.

ndng28, 2023-04-29:
Looks like this issue is resolved for me. Not sure what helped but here are the sequence of events...

- Upgraded from 3.11.x to 3.12.0 and started seeing prompts for filament change during print.
- Cleaned filament sensor as advised but did not help
- Downgraded to 3.11.x and the issue was resolved
- Upgraded again to 3.12.2 and so far no issue with over 100 hours of print!

frippe75, 2023-05-25:
I'm now on prusa3d_fw_MK3_3_12_2_5713.hex after having exactly this issue with my earlier firmware (which I cannot determine).

But I happened the last three times on the old firmware at the exact same place 1h into the print. Upgraded and now it happened even earlier.

Prusa-Support, 2023-05-26:
This issue is recently becoming popular on GitHub and surely we won't ignore this, but so far it seems hardly reproducible.

Our Technical Support though has sporadically seen similar reports in the past, all time, with all firmware versions.
This is usually an electronic problem with the wiring or a mechanical problem around the triggering mechanism.
For this reason, I'd like to stress to all users to double-check these aspects in the first place and possibly continue trying alternating firmware versions.

Thank you for keeping on reporting here and on the other specular issue ("equally in importance" hence not closed as duplicate) https://github.com/prusa3d/Prusa-Firmware/issues/4061.

Unfortunately, we couldn't reproduce and confirm a bug yet. The relevant portion of the firmware hasn't been edited and apparently the serial logs wouldn't help either.

However, we acknowledge the existence of an unconfirmed bug based on the number of recent reports and we have reason to believe that if such a bug is there, it won't persist in the future 3.13 firmware. Please check this especially informative comment https://github.com/prusa3d/Prusa-Firmware/issues/4061#issuecomment-1501790648.

Michele Moramarco
Prusa Research

benderl, 2023-06-25:
Yesterday I noticed this behaviour on my stock MK3 (no S/+). Finally I found this issue and today I downgraded to 3.11.0. Unfortunately the printer keeps stopping for filament (un-)load. I never had problems with the filament sensor, but disabling it solved it. I have some more prints to do and will check the sensor before the next one.

Maybe this is another issue and there is dust on the optical sensor. But at least I got the same error with 3.12 and 3.11 firmware versions.

mozgy, 2023-06-28:
I'll jump on #meetoo bandwagon, stock MK3, purchased the very first day, still works fine. PLA here, SD printing. Unload/load never happens on the same spot even with the same gcode. Tried dusting, changed nozzle, no help, still random spot unload/load.
Backpedaling to 3.11 that same gcode works fine.

tideline3d, 2023-07-17:
24 printer farm here.   I'd upgraded a few printers to 3.12.2 as I repaired hotends over the last few months with no issues other than it catching some flaky thermistors.   

Two days ago I upgraded the whole farm.   2 of 24 are now doing this uncommanded filament change on gcode that I've been running unchanged for 1+ year.   Both printers run 12-14hr/day for the last year printing this exact gcode (yay Prusa) with zero problems of this nature.    I keep my gcode in a Git repo for reference, it is 100% the same as before.

All printers are running Octoprint on RPI4/2gb (and have been for 1-3 years depending on the printer, these two are mid 2021 purchased printers).  

Please reach out if there's any data I can provide with the size of the farm (and most working fine), happy to help.  I see you've re-written this for 3.13 and that seems to be the next milestone in this issue.  

For my own notes, it was printer C1 and C3.

DRracer, 2023-07-17:
To all who are experiencing the fsensor issues - could you please try flashing the latest FW from branch MK3_3.13.0 and report back if the issues still persists.

@tideline3d If the 2 printers of yours keep on doing bogus runouts - that may be the best test case to verify on.

tideline3d, 2023-07-17:
Will do. Let me run them a couple more days to make sure it’s somewhat reproducible then I’ll upgrade one of them.

tideline3d, 2023-07-24:
FWIW this has not been reproducible on those two printers. It happened 2 or 3 times right after the firmware update and they've been running smooth for a few days since.

DRracer, 2023-07-28:
@tideline3d if I understood correctly, you updated to 3.13, bogus fsensor triggers occurred 2 times, but it didn't happen ever since, right?

tideline3d, 2023-07-28:
@DRracer I have not done 3.13 yet. I did 3.12 and printers C1 and C3 did this unnecessary filament change on the next print after the update. 

I have not seen those two do it again so I haven't tried 3.13.  I'm not a fan of changing things if I can't prove it fixed something. 

I had one more printer do this since, E2.  But have yet to have 'reliable' failures to test against. 

All 24 printers are running 12-16hrs a day for reference.

DRracer, 2023-07-28:
@tideline3d fair enough, once you flash 3.13 at some point and the printers still cause bogus runouts, please report.

tideline3d, 2023-07-28:
@DRracer I spoke too soon.  C3(repeat offender) and now E1 (new offender) did this when I went to the shop today.   I was there while it happened and snagged a video from my cameras.   I know it's not terribly useful, but proof that it does exist.  I reloaded it and it went on to print just fine.   I wiggled the IR sensor cable a bit to see if I could reproduce it that way, but no luck.   

https://youtu.be/Ymc4aP5aSaw

I looked through the 3.13 changes and don't see anything that stands out as a logic change to the filament runout.   Happy to try it though, just want to make sure I'm running what you expect.   Here's the branch I was looking at https://github.com/prusa3d/Prusa-Firmware/compare/MK3...MK3_3.13.0

tideline3d, 2023-08-07:
Just an update here, none of these are consistent.  I've had one more do this since the last post, but none of the 24 are doing it in any kind of repeatable pattern.  I'll keep watching and update if I find anything useful.

DRracer, 2023-08-08:
@tideline3d thanks for updating this issue. Currently, we are neither sure where the occasional issues are coming from nor how to reproduce them. We suspect the ADC as this code has been refactored for the thermal model data acquisition.

EveryAmp, 2023-08-25:
We're also seeing this on 1-3 printers per day out of 90 MKS/+. It first started with 3.12.2 and has persisted with 3.13. As best as we can tell it's random. Different printers every day running various gcodes stopping in random places. Let us know if there's anything we can do to help track it down.

krite, 2023-10-30:
Having no real issues with PLA or PETG, trying ABS again (was quite some time ago) this exact behaviour shows directly after the wipe-string. Then a new change after the 'outline'. Then again after the first 1.5 perimeter is laid down. Each print, repeatable. I can reproduce for this print, 5 times in a row now. This print is thus not possible to get even to start before the filament change occurs. Also, turning off the filament sensor is not possible. The outer-settings-menu shows the sensor ass 'on' but going into the sub-menu shows it as 'off'. It cannot be changed. The status remains the same regardless. Tried 3.12 and latest 3.13.2. Not nice. Not nice at all. MK3 stock. Prusa ABS black.

krite, 2023-10-31:
Downgraded to 3.11.0, the menu can now accurately show the actual status of the filament-sensor (it was on). The submenu works, it can be turned off. Also, printing now works with ABS again. Please fix.

goobags, 2023-12-18:
I am also seeing this bug, although on 3.13.2 firmware (technically Bondtechs firmware. Never occurred on previous firmware which from memory was pre-3.11 as I don't always keep firmware updated. Seems completely random, some prints will do it every layer and some will only do it once, or not even at all. Downgrading to 3.11 to see how I go.

closed 2024-02-26 by github-actions[bot]