prusa3d/Prusa-Firmware #3882: [BUG]<Enter comprehensive title>

4sStylZ, 2023-01-04:
**Printer type** - Mini +
**Printer firmware version** -  mini_release_boot_update_pre_4.4_4.4.0+4180.bbf

**MMU upgrade** - no
**MMU upgrade firmware version** - no

**SD card or USB/Octoprint**
SD card only

**Describe the bug**
Note that this was working well just before my firmware update. But since I use a new gcode, I can't only follow this path : 
**On every filament change**, I get a message that completly block the printer : « Mot de passe » which is the french word for « Password ». 
  
**To Reproduce**
I get it on every filament change.
  
**Expected behavior**
 When loading the new filament, I don't want this message to appear.
  
**G-code**
[Top key cover Bicolore PLA ASA_0.15mm_ASA_MINI_31m gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/10341292/Top.key.cover.Bicolore.PLA.ASA_0.15mm_ASA_MINI_31m.gcode.txt)

**Image**
[image]

What I searched for now : 

1. I didn't found any « Mot de passe » or « Password » thing in the gcode with my text editor.
1. I searched for the word « Mot de passe. » in the french translation file named `Firmware_fr.po`. And I didn't found it.
1. I restarted multiple time the printer and prusa printer.
1. I disabled disable all the new wifi and network stuff on the Mini +

Since it's not in the gcode, and not in the translation file, is it a good idea to assume that at least the message is stored somewhere on the prusa memory ?

Thank's.

sarusani, 2023-01-04:
This is a bad translation in:
/src/lang/po/fr/Prusa-Firmware-Buddy_fr.po (Line 991)

The original english message is: "Please remove filament from filament sensor."

4sStylZ, 2023-01-04:
You're completly right and saved me !

This new procedure was new to me, and I created the buddy issue : 
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2883

closed 2023-01-04 by 4sStylZ