prusa3d/Prusa-Firmware #3888: [BUG] Pronterface bug when starting again a finished print.

laskr1999, 2023-01-10:
Printer: MK3S+
FW: v3.12.0-RC2
MMU: No 
SD Card was in the printer, but printed from pronterface

**Describe the bug**
 If i print something with pronterface, and it is finished, if I want to reprint that thing again, last line error comes up spamming in the lines. 
  
**To Reproduce**
Print something through pronterface, then after finish, just hit again the print button
  
**Expected behavior**
Just start to print as before. 
  
**G-code**
  Not matter, it does with everything, but attached the gcode renamed to png. 

[image]

laskr1999, 2023-01-11:
In the lastest live build its still there

wavexx, 2023-01-15:
@laskr1999 that seems like a bug that needs to be filed to https://github.com/kliment/Printrun/issues/

laskr1999, 2023-01-27:
@wavexx are you sure? Noticed that after a print, I get a full nx buffer or like that. Before this FW it was perfect.

wavexx, 2023-01-27:
Oh, that might be relevant actually, it might be dropping the first M110 for whatever reason.

@leptun can you think of why rx could be full in 3.12?

@laskr1999 can you do a short print, and when hitting reprint save all the output you get in the log? Looks like the gcode is irrelevant here, so we need to see exactly what is sent just before and when you reprint.

laskr1999, 2023-01-27:
@wavexx how to make the log? Doing a print right now, maybe tomorrow morning can do a fast print, if I have time, but surely next monday is when I can do testing. 

At least what i noticed:
Printing, its done, at the end it says rx full. 
I hit restart/print another code and just spamming that thing, and in the picture where it shows the print it just gets green very fast line by line. 
I disconnect/reconnect and it's good. 

Cannot really do reset, because after that, it stucks at 3 dot, and not showing FW version. But a power cycle makes wonders and do reset properly. RC and stable is the same for this problem.

laskr1999, 2023-01-28:
Ok seems like that rx buffer is sometimes goes full. Mainly just that resend bug. Are there any newer version from it? I use the one that comes with ps2.5
[image]

Prusa-Support, 2023-02-23:
> how to make the log? 

Once the problem occurs, please copy the text in the [Info Panel (10)](https://help.prusa3d.com/article/pronterface-and-usb-cable_2222) starting several lines before the print actually ends.
You can paste it into a simple .txt file and send it over here.

I'm not aware of progress in regards to this issue, but [FW 3.12.2](https://github.com/prusa3d/Prusa-Firmware/releases) should be published sometime soon. Please check back and give it a try in a few days.

Michele Moramarco
Prusa Research

digitalcraft, 2023-05-12:
issue has been open for a while on both projects.

laskr1999, 2023-05-15:
> Once the problem occurs, please copy the text in the [Info Panel (10)](https://help.prusa3d.com/article/pronterface-and-usb-cable_2222) starting several lines before the print actually ends. You can [...]

Hello!
Latest FW still not ok. Here is 2 ways of the problem attached. First if i press disconnect to stop it, second if I let it run through.

If i let it run through, it will fastly get green line by line, then print ended. If i disconnect it it will just stop. Same gcode etc.
Txt attached. 
[fw probrem.txt](https://github.com/prusa3d/Prusa-Firmware/files/11480747/fw.probrem.txt)

laskr1999, 2023-08-02:
Latest firmware still happens, updated everything(slicer, printer fw, pronterface). Its annoying that we cannot use it correctly, because we have to disconnect/reconnect or reset the printer itself to start a new print.

closed 2023-11-21 by github-actions[bot]