prusa3d/Prusa-Firmware #3889: [ENHANCEMENT] The new Nozzle Change feature should max out the z height

laskr1999, 2023-01-10:
Printer:MK3S+
MMU: no

The new nozzle change feature is good, but the problem is, the bed nearly stays under the nozzle. When someone(like me) uses a screwdriver like socket I can't fit it under the nozzle just in a steep angle because of not maxed out z height+bed is under it.
  
**Describe the solution you'd like**
  I think the bed should go to the zero position to the back and the Z to the max 210mm so anything can fit under the nozzle very well/make it configurable so peoples can adjust to their wrenches.

3d-gussner, 2023-01-12:
@laskr1999 Thanks for the feed back.

> I think the bed should go to the zero position
Agreed on the Z height and Y as far back, but I still would keep the X pos in the middle to have access from all sides.

3d-gussner, 2023-01-13:
@laskr1999 Thanks for the request. It will be implemented in next release see https://github.com/prusa3d/Prusa-Firmware/pull/3901 and https://github.com/prusa3d/Prusa-Firmware/pull/3902

Please close this issue / feature request.

3d-gussner, 2023-01-16:
Closing as the enhancement has been merged.
Thanks @laskr1999 again for the feedback :hugs:

closed 2023-01-16 by 3d-gussner