prusa3d/Prusa-Firmware #3891: [ENHANCEMENT]3.12.0: Make thermal model disabled also prevent calibration requirement

arekm, 2023-01-11:
**Printer type**: MK3S+

**Is your enhancement related to a problem? Please describe.**

With new firmware 3.12 introducing thermal model (TM) a problem comes up with modded printers (like with revo hotend added)
which requires TM to be disabled.

That's easy to do via "M310 S0 + M500".

The problem is that doesn't disable thermal calibration requirement after firmware upgrade etc.

Example scenario:
- printer with revo
- 3.12.0 RC2 firmware; M310 S0 + M500 done
- upgrade to final 3.12.0, printer boots and pauses ("Recv: echo:busy: paused for user") requiring manual intervention (on another floor in my case) to be able to control printer via serial again. Printer wanted to do TM calibration (which won't work on revo anyway).

**Solution**

So maybe firmware should not only check EEPROM bit for "is thermal model calibrated" but also should check if user has TM enabled?

That would prevent asking user to do calibration which won't work with revo and as TM is disabled anyway on such printer.

arekm, 2023-01-11:
Hi  @wavexx .

"This is because RC2 still didn't have an enforced calibration workflow as part of the wizard. This shouldn't be the case for all future versions." (from https://github.com/prusa3d/Prusa-Firmware/issues/3890#issuecomment-1378998100).

Ok, I started the wizard, it failed (due to revo). How now to tell it to never even try again? (until M310 is enabled). It seems to insist that TM wizard needs to succeed.

So now every printer boot or octoprint connect via serial - printers stops ("Recv: echo:busy: paused for user") waiting for lcd button press due to "Temp model not calibrated yet" (which means trip to another floor).

M310 S0 + M500 changes nothing. It still wants button press for "Temp model not calibrated yet" confirmation every boot/serial connect.

wavexx, 2023-01-11:
The dumbest way is probably `M310 A F0` right now (this should succeed and clear the calibration required as a result).

`D3 Ax0ca6 X1f` will also manually mark it as done by manually updating the eeprom.

Neither a nice solutions, I know. The calibration status is now set a bit mask, but there's no command to set the individual bits (which IMHO is the way to go). `D10` allows this for XYZ, maybe we should extend this one.

arekm, 2023-01-11:
"D3 Ax0ca6 X1f" worked (from my understanding it sets bits for selftest, xyz calibration, z calibration, TM calibration and 1 layer calibration as "done").

FrHePo, 2023-01-14:
I sent D3 Ax0ca6 X1f and got this back: *** Unknown syntax: D3 Ax0ca6 X1f. Still getting "Temp model not calibrated yet".

FrHePo, 2023-01-14:
Downloaded newest Pronterface and tried again:
>>> D3 Ax0ca6 X1f
SENDING:D3 AX0CA6 X1F
D3 - Read/Write EEPROM
0 bytes written to EEPROM at address 0x0000

BUT D3 didn't work, so still getting "Temp model not calibrated yet" at start-up!

gudnimg, 2023-01-14:
> SENDING:D3 AX0CA6 X1F

I suspect the issue here is Pronterface is sending the A and X parameter arguments with capitalized letters. AX0CA6 and X1F should be AX0ca6 and X1f since the code is case sensitive.

The firmware sees AX0CA6. It interprets it as A parameter is empty, X contains 0

FrHePo, 2023-01-14:
FW 3.12.0. Printrun 2.0.0rc8 (Github). Newest Pronterface as downloaded from Prusa doesn't recognize D3 Ax0ca6 X1f

FrHePo, 2023-01-14:
Would Octoprint do a better job?

gudnimg, 2023-01-14:
@FrHePo I would try Octoprint if you can. Pronterface won't work because it looks like it is converting the G-code command in uppercase.

Arduino IDE's serial monitor also works and is easy to use. Just don't forget to set the baudrate to 115200.

[image]

FrHePo, 2023-01-14:
https://github.com/kliment/Printrun/issues/216

wavexx, 2023-01-15:
@FrHePo who-ho.. auto-capitalization. I tried this now with `pronsole` and indeed, it does that too.

> The command line tries to be helpfu by capitalizing the line if the initial m/g character is lowercase.

that doesn't seem to be the case... it autocapitalizes everything. Is this a bug?

FrHePo, 2023-01-15:
It should be possible to switch autocapitalization on and off. I think this is a due enhancement.

wavexx, 2023-01-15:
Looks like that using the "raw" `@` prefix at least doesn't capitalize (a prefix I keep forgetting about in printun).

FrHePo, 2023-01-15:
> @FrHePo I would try Octoprint if you can. Pronterface won't work because it looks like it is converting the G-code command in uppercase.
> 
> Arduino IDE's serial monitor also works and is easy to use. Just don't forget to set the baudrate to 115200.
> 
> [image]

I'm not familiar with Arduino :-((

FrHePo, 2023-01-15:
> Looks like that using the "raw" `@` prefix at least doesn't capitalize (a prefix I keep forgetting about in printun).

Do you mean to enter  @D3 Ax0ca6 X1f ?

wavexx, 2023-01-15:
Yes, typing `@D3 Ax0ca6 X1f` at the prompt should do the right thing

FrHePo, 2023-01-15:
Worked, thanks!
>>>@D3 Ax0ca6 X1f
SENDING:D3 Ax0ca6 X1f
D3 - Read/Write EEPROM
1 bytes written to EEPROM at address 0x0ca6
000ca6  1f

PR #3916 (2023-01-16): wizard: Do not complain about missing calibration if model is disabled
This fixes #3891 without having to set the calibration bit, meaning that if the model is later re-enabled without a real calibration, a prompt is shown as expected.
commit 12f5bd36 (2023-01-16): wizard: Do not complain about missing calibration if model is disabled

This fixes #3891 without having to set the calibration bit, meaning
that if the model is later re-enabled without a real calibration, a
prompt is shown as expected.
commit 7adfd0c6 (2023-01-16): wizard: Do not complain about missing calibration if model is disabled

This fixes #3891 without having to set the calibration bit, meaning
that if the model is later re-enabled without a real calibration, a
prompt is shown as expected.
closed 2023-01-16 by DRracer