prusa3d/Prusa-Firmware #3893: [BUG] Hotend fan keeps running after failed Thermal Model calibration

gregsaun, 2023-01-11:
**Printer type** - Prusa MK3S+ with Bear customs and E3D Revo Micro hotend
**Printer firmware version** - 3.12.0 Final
**MMU upgrade** - None

**Describe the bug**
When I do the calibration for the new Thermal Model it fails due to the Revo hotend with a THERMAL ANOMALY error. The hotend fan is starting to rotated and doesn't stop even if the temperature goes bellow 50°C.
  
**To Reproduce**
1. Power on the printer (hotend is at 21°C)
2. LCD menu -> Calibration -> Temp Model cal. => THERMAL ANOMALY is raised due to the Revo hotend. The hotend fan start spinning while it still under 50°C (same behavior if it goes over 50°C)
3. LCD menu -> Clear TM error => hotend fan is still spinning (hotend is under 50°C)
4. LCD menu -> Preheat -> Cool down => hotend fan is still spinning (hotend is still under 50°C)
5. After few minutes => hotend at 23°C and hotend fan still spinning
6. Power off the printer and then power on => fan stops spinning
  
**Expected behavior**
Once the hotend is bellow 50°C the fan should stop.
  

Thanks for the help

Aldebaran91, 2023-01-12:
3.12 doesn't work with E3D Revo Micro hotend. I just downgraded my MK3S+. I will wait till there is a usable version.

arekm, 2023-01-13:
It works but there are (configuration) dragons: https://github.com/prusa3d/Prusa-Firmware/issues/3636#issuecomment-1380243362

matthew-humphrey, 2023-01-13:
Not a bug. Once it sees something unexpected, it assume something is wrong, and leaves the heater off and the fan on as a safety measure. Klipper does the same thing when it detects a hotend or bed heater anomaly.

wavexx, 2023-01-15:
Bug in this case. We do support resuming from a thermal model error, just in case the thermal error was transient.

In that case restoring the previous hotend fan state is ok (we only allow to resume if the error is gone anyway - you cannot resume as long as the error state is triggered).

This is new, since spinning the fans at full speed as an extra precaution wasn't done previously.

wavexx, 2023-01-15:
As a food-for-thought, I was thinking that spinning fans works if we didn't catch fire *yet*. We would probably be fueling the flame otherwise ;)).

gregsaun, 2023-01-15:
> In that case restoring the previous hotend fan state is ok (we only allow to resume if the error is gone anyway - you cannot resume as long as the error state is triggered).
This was my thought as well, good thanks 👍

> We would probably be fueling the flame otherwise ;)).
Could be yes, probably not for too long 😁

matthew-humphrey, 2023-01-15:
> Bug in this case. We do support resuming from a thermal model error, just in case the thermal error was transient.
>
> In that case restoring the previous hotend fan state is ok (we only allow to [...]

It's kind of a play on the old "halt and catch fire" from the early microprocessor days. 😆

On a serious note, since 3.12 has since been released, is an adjustment to the thermal model to handle these modded printers (for example with a alternate extruder with Revo) still planned? If so, will it just be Revo, or do you plan to support other extruder+hotend combos?

wavexx, 2023-01-15:
> It's kind of a play on the old "halt and catch fire" from the early microprocessor days.

Still a proud owner of an SGI Octane 2, although I don't remember if that one has the halt_and_catch_fire flag ;) - sadly didn't turn it on in a few years

> On a serious note, since 3.12 has since been released, is an adjustment to the thermal model to handle these modded printers (for example with a alternate extruder with Revo) still planned? If so, will it just be Revo, or do you plan to support other extruder+hotend combos?

Not sure if the tweak for Revo makes sense for 3.12 or it will be rolled up to 3.13. However, other hotends should work. If you have examples that don't work _right now_ it would be helpful to know, since the most annoying part for the model is the autocalibration part, and any new variable there is a true PITA to take into account for upgrades. It's much better to know what's not working from the get go.

matthew-humphrey, 2023-01-15:
> If you have examples that don't work right now it would be helpful to know, since the most annoying part for the model is the autocalibration part, and any new variable there is a true PITA to take into account for upgrades.

I don't as I am not using a modded MK3 anymore. But my sense is the other commonly used hotends (I.e. Phaetus Dragon ST) will have a very similar thermal behavior to the V6.

Thanks for the prompt replies. I'm continually amazed at how much you guys manage to pack into the limited flash RAM on the atmega2560. Can't wait to see what you will be able to achieve once you've migrated fully to 32b.

closed 2023-01-16 by DRracer

adamdport, 2023-02-02:
I upgraded my mk3s firmware on January 26th. On the 27th I noticed that my prints were very droopy, drooping on overhangs and bridges. I didn't make the connection that I'd just upgraded my firmware and so I ordered a new print fan. It arrived today and I just installed it but it didn't improve my prints. I then remembered that I'd just upgraded my firmware and decided to downgrade it and try again, but to no avail–my prints are still droopy on 3.11.0, despite having printed the exact same gcode hundreds of times over the past year without any issues. I can see that it's spinning during the print, though I don't have any way of comparing how fast it's spinning to how fast it _used to_ spin.

Is there something that this firmware may have tweaked that wouldn't be removed by downgrading? Something I need to recalibrate for example? Open to any ideas, thanks.

commit 16922c26 (2023-01-16): TM: Restore extruder autofan state when stopping/resuming

During thermal errors all fans are set to full speed.

When the print is resumed or stopped *and* the thermal error is gone,
also restore the autofan state.

Fixes #3893
commit 7529652d (2023-01-16): TM: Restore extruder autofan state when stopping/resuming

During thermal errors all fans are set to full speed.

When the print is resumed or stopped *and* the thermal error is gone,
also restore the autofan state.

Fixes #3893