prusa3d/Prusa-Firmware #3894: [ENHANCEMENT] M310 S0 should disable 'Temp model not calibrated yet' message on boot

darraghbr, 2023-01-11:
With version 3.12 if the thermal model is not calibrated the printer will display this message upon boot. 
  
[image]

Unfortunately this message prevents PrusaLink from starting and so if you start the printer remotely (smart plug etc) you then need to go to the printer, clear this message and then the printer will be operable remotely. 

**Describe the solution you'd like**
If M310 is set to S0 this should disable this message. Especially currently when it is not feasible to calibrate the thermal model for any type of alternative hotend (Revo etc).

arekm, 2023-01-11:
See https://github.com/prusa3d/Prusa-Firmware/issues/3891
Duplicate.

darraghbr, 2023-01-11:
I did read that issue and I thought you were talking about the prompt after firmware upgrade to calibrate the thermal model, rather than the reminder upon every boot. The wording is quite unclear.

closed 2023-01-11 by darraghbr

FrHePo, 2023-01-14:
I tried M310 S0 + M500 but I'm still getting "Temp model not calibrated yet" at start-up!
>>> M310 S0
SENDING:M310 S0
>>> m500
SENDING:M500
echo:Settings Stored

darraghbr, 2023-01-15:
> I tried M310 S0 + M500 but I'm still getting "Temp model not calibrated yet" at start-up!
> 
> > > > M310 S0
> > > > SENDING:M310 S0
> > > > m500
> > > > SENDING:M500
> > > > echo:Settings Stored

You should try this:

https://github.com/prusa3d/Prusa-Firmware/issues/3891#issuecomment-1379287828

It looks like you have to do it through Octoprint rather than Pronterface because of the way Pronterface interprets your input.