prusa3d/Prusa-Firmware #3895: [BUG] MMU2S/printer can't progress following filament failure intervention

3dprintandpaint, 2023-01-12:
*** This is a copy of #3742, which is present in the Beta and describes the problem perfectly.  The issues persists in the final release making the MMU unusable after any attempted intervention ***

Printer type -MK3S+
Printer firmware version - 3.12.0

MMU upgrade - MMU2S
MMU upgrade firmware version - 1.0.6

USB/Octoprint

Describe the bug
When MMU2S is in failure state (e.g., filamant is not loaded, or out-of-filament), then printer displays error message containing like "press button on MMU unit to continue". Anyway pressing any button on MMU2S unit or recovering the MMU2S failure (by loading the filament) is not read out by printer and printer is stuck forever, so only possible solution is to reset it.

To Reproduce
Use printer menu to load filament but do not insert the filament to MMU2S. After a while printer displays error message and it is not possible to recover printer by loading the filament (using MMU2S buttons). Printer is stuck and the only solution is to reset printer.

Expected behavior
When MMU2S failure is resolved (e.g., by loading the filament using MMU2S buttons) printer shall continue.

G-code
N/A

Video
NONE

3d-gussner, 2023-01-12:
Thanks for reporting this issue. We are looking into it.

3d-gussner, 2023-01-13:
@3dprintandpaint Can you please try this firmware?
   
[prusa3d_fw_MK3S_MK3S+_t3_12_0-1_5673.hex.zip](https://github.com/prusa3d/Prusa-Firmware/files/10413481/prusa3d_fw_MK3S_MK3S%2B_t3_12_0-1_5673.hex.zip)

If it is working please report back and close this issue?

3d-gussner, 2023-01-16:
Closing issue as the fix has been merged for FW 3.12.1

closed 2023-01-16 by 3d-gussner