prusa3d/Prusa-Firmware #3896: [BUG]<German language copy fails>

wseifert, 2023-01-13:
**Printer type** - MK3S
**Printer firmware version** -  3.12

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**SD card or USB/Octoprint**
  USB/Octoprint

**Describe the bug**
  After flashing the new FW 3.12 Printer asks for Language to copy; selecting German the printer asks again for language afte copy process
  
**To Reproduce**
  Flash printer with FW 3.12 ans select German language
  
**Expected behavior**
  language copy process finishes sucessfull
  
**G-code**

**Video**

arekm, 2023-01-13:
What way you used to flash firmware?

wseifert, 2023-01-13:
Via USB & PrusaSlicer 2.5. Flashing with 3.11 worked fine except a error message in PrusaSlicer at end of flash process (I do not remember exact what message, connection to MK3S was interupted at end or so, MK3S had 3.11 FW and worked fine). So I switched of and on power, reflashed again with 3.12 and now it worked without error. Temp Model calibration worked with one error message somewhere after 5 minutes (temp anomalie detected) but finished and after power off and on again MK3S has FW 3.12 and is working fine.

closed 2023-01-13 by wseifert