prusa3d/Prusa-Firmware #3898: MK3S+ on 3.12.0 occasionally does a continuous beep when hitting MINTEMP

knmorgan, 2023-01-13:
**Printer type** - MK3S+
**Printer firmware version** -  3.12.0

**MMU upgrade** - N/A
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
Octoprint via RPi Zero 2 connected directly to the EINSY board, but no active print when experiencing this bug.

**Describe the bug**
The printer occasionally does a continuous beep when hitting MINTEMP. The LCD also goes blank; this requires a reset or a power cycle to resolve. Ordinarily when it hits MINTEMP on 3.12.0 it does a short-ish beep (maybe 3 seconds?) and I do not recall hearing any beep at all prior to 3.12.0. My prior firmware version was 3.10.1.

When I reset the printer (not captured on video, unfortunately) it did the usual short-ish beep and showed the bed at 8°C and the hotend at 10°C.
  
**To Reproduce**
1. Place printer in a garage during the winter.
2. Wait for the garage to get below 10°C.
3. This behavior is inconsistent; I do not know of any variables to describe the difference in behavior.

**Expected behavior**
No continuous beep. Ideally, no beep at all.
  
**G-code**
N/A

**Video**

https://user-images.githubusercontent.com/538102/212331597-57ec3f79-097f-4895-a441-78814f0ec368.mov

knmorgan, 2023-01-13:
It may also be worth noting that after hitting MINTEMP the hotend fan runs constantly. I’m fairly certain this was not the case prior to 3.12.0.

gudnimg, 2023-01-13:
You report the LCD goes blank… that is definately abnormal. I’ll try taking a look at this over the weekend and see if I can reproduce and fix the issue.

gudnimg, 2023-01-13:
When you get a endless beep. What is the printer doing at that moment? Is there a full-screen message showing on the LCD? The only way I can see the buzzer not stopping to beep is if the printer is doing something else, like waiting for user input.

Serial logs would be very helpful since I am not able to reproduce this simply by triggering MINTEMP error.

What's the status of your Temperature model? On my printer it has no been calibrated yet. Is yours calibrated or disabled?

gudnimg, 2023-01-13:
Also good info to know regarding the LCD: Do you have any languages enabled or do you use the English only firmware build?

knmorgan, 2023-01-13:
> When you get a endless beep. What is the printer doing at that moment? Is there a full-screen message showing on the LCD? The only way I can see the buzzer not stopping to beep is if the printer is doing something else, like waiting for user input.

The printer was idle doing nothing. I heard a loud beep from inside my house and it took a few minutes to discover it was the printer in the garage. When I went out the LCD was blank with the exception of the icon in the bottom-right. You can see this in the attached video. I had not interacted with the printer at all when I recorded the video, and it had been idle for many hours prior.

> Serial logs would be very helpful since I am not able to reproduce this simply by triggering MINTEMP error.

Can you let me know how to provide serial logs? This does not happen reliably when triggering MINTEMP for me either. I regularly encounter MINTEMP in the winter and I believe I've triggered it three times since upgrading to 3.12.0. This bug (continuous beep, blank LCD) only happened one of those three times.

> What's the status of your Temperature model? On my printer it has no been calibrated yet. Is yours calibrated or disabled?

I believe it calibrated automatically when I first upgraded to 3.12.0 but if you let me know how to check this I can verify.

> Also good info to know regarding the LCD: Do you have any languages enabled or do you use the English only firmware build?

English only.

knmorgan, 2023-01-14:
This happened again tonight; I'm uploading another video. The LCD was not blank this time, but it's now telling me to calibrate the printer. Very confused because I've never had to do this before.

https://user-images.githubusercontent.com/538102/212449227-8b65fb04-a3ea-4286-8566-59b161ca895d.mov

gudnimg, 2023-01-14:
In 99% of boot-ups the printer just lets me know the thermal model is not calibrated yet `"Temp model not calibrated yet."` . But 3 times I randomly saw the firmware asked me to calibrate the thermal model and if I did not select anything it would timeout and go ahead with the calibration anyway (I canceled by rebooting). I have not looked into this behavior in more detail to explain it.

knmorgan, 2023-01-14:
@gudnimg If there are any logs that would be useful to gather let me know. I will probably revert back to 3.10.1 otherwise.

gudnimg, 2023-01-14:
Regarding the LCD blank issue, logs will certainly may be very helpful for me to determine where the issue is coming from. You should be able to use Octoprint to capture the logs from the printer. I'm not a Octoprint user myself, I'll post some instructions if I find any shortly (they've probably been explained before here on Github).

But regarding MINTEMP I don't think there is anything wrong there except maybe the behavior of the buzzer beeps. I can't promise any fix, but this needs some more discussion.

gudnimg, 2023-01-14:
@knmorgan https://docs.octoprint.org/en/master/bundledplugins/logging.html does this help? Seems to be a plugin from Octoprint.

gudnimg, 2023-01-14:
@knmorgan Prusa has some information about this here (bottom of page). You will need to enable serial logging as shown in the video. https://help.prusa3d.com/article/crash-dump_364959

wavexx, 2023-01-15:
This is likely the thermal model hitting thermistor instability as it goes below MINTEMP. There's no special handling for extra thermistor noise/unavailability when it dips below MINTEMP, so I assume it just either triggers randomly or enters full error state at some point and issues a ThermalStop (which is what is causing all fans to spin).

I'm unsure whether we can make use of model checking below MINTEMP. Even if we disable model check for cooldown only with heaters off, the noise will be sufficient to result in the same while heating up until we pass some threshold.

Seems to me the only thing we can do is to disable it if both einsy/hotend read low and heaters are off.
But _then_ you really want some proper mechanism to raise above MINTEMP safely.

knmorgan, 2023-01-15:
@wavexx Your comment does not address the bug raised in this issue. I'm aware of the difficulties around differentiating between a very cold temperature and a disconnected thermistor, but I really only wanted the endless beep addressed in this issue. I've captured two separate instances of this on video.

If we're speaking more broadly about how the MK3S+ behaves in cold environments, I'm really surprised the new thermal model isn't able to more intelligently distinguish between a bad thermistor and a cold environment. This is the single most frustrating shortcoming of the MK3S+ for me and I've seen a fair bit of discussion around it. Even if you ignore the Einsy temperature, presumably a slow, continuous temperature decline over time on the hotend/bed looks different than a faulty thermistor. This seems like a much simpler problem to solve than the blob detection mentioned on the Prusa blog.

gudnimg, 2023-01-22:
Randomly hit this endless beep when I was testing 3.12.1 today.

[image]

knmorgan, 2023-01-30:
@gudnimg I have not had the opportunity yet to retrieve logs, but just wanted to point out this still happens regularly for me. Anecdotally, it seems every time I've had the printer on for an extended period of time (several hours at least) before hitting MINTEMP it seems to trigger the endless beep. When I restart the printer and it immediately triggers MINTEMP it does a relatively short beep (~3 seconds). Most of the time this happens the LCD is not blank like I initially reported.

Prusa-Support, 2023-02-25:
Sorry for tackling in just now.

I'd like to report that I recall observing a similar issue with some customers in the past. I may seem to recall it was caused by a MAXTEMP error at the startup, while the Wizard was being automatically starting.
I believe it was before the FW 3.12.0 release.

Based on my wild guess, a critical MAXTEMP error is being triggered while the printer was prompting a specific wizard/message, like "thermal model/printer not calibrated".

I see how a partially damaged thermistor could trigger erratic MINTEMP errors, and in specific conditions can be subject to short circuits. It is a fact that a shorting thermistor can report a too-low resistance, hence the printer will think that the temperature is very high (MAXTEMP error).

My interest back then was replacing the thermistor to prevent the problem.
You may be interested in reproducing the problem instead.
If you intentionally short the thermistor while a wizard is starting, I think you can reproduce the issue.

Michele Moramarco
Prusa Research

gisforgirard, 2023-03-17:
https://github.com/prusa3d/Prusa-Firmware/issues/3053#issuecomment-1473092044

Prusa-Support, 2023-04-20:
The loud beeping behavior should be improved by PR #4119.
Thanks for reporting and bearing with us. 

Michele Moramarco
Prusa Research

Prusa-Support, 2023-08-08:
This issue will be closed as fixed now.
Please test the most recent firmware release where the above-mentioned fix is supposedly merged.

Michele Moramarco
Prusa Research