prusa3d/Prusa-Firmware #3912: [BUG]<Enter comprehensive title>

gold1640, 2023-01-14:
Printer is i3Mk3S, assembled by Prusa in 2020, no modifications. 

Installed v3.12. "Thermal Anomaly" during calibration. 

The first part of the calibration seems to work:-

TM: initial C est. (target seems to be 230C)

TM: cool down <50C (cooling down to <50C)

TM: refine C est. (target again seems to be 230C)

As the temperature reaches 224-225C the fan goes full speed and the temperature starts to fall: Thermal Anomaly on screen and alarm beeping.

Re-installed v3.11.

--
Ian Harris

fink-at-trmc-dk, 2023-01-15:
Same here. Rolling back to 3.11
//Fink

josh3io, 2023-01-16:
It took me three attempts to get a successful calibration without TA error. Of course I got M112 on the first layer of the first print post-calibration. Agree with others that the thermal model feature requires more development. It seems to be even more sensitive than the previous firmware version to minor spikes in the temperature control. What used to result in recovery and print continuing successfully now results in complete failure of the system.

PowerChaos, 2023-01-16:
same problem for me 
only a bit more extreme

at 18 degrees nozzle it goes directly into  "Thermal Anomaly"
every single time after 1 or 2 seconds it abort the "calibration"
need to roll back as else i can not even start a single print ... 

Sidenote , using a Revo nozzle ? maybe it heats up to fast ?
but works fine on 3.11.1

DRracer, 2023-01-17:
If you experience thermal anomalies while doing the calibration, going back to 3.11 is the worst option you can do. 
Please read the release notes about the purpose of the thermal model safety check **again** and **think twice before reverting to the previous firmware**, especially in case of stock printers.

josh3io, 2023-01-17:
1. I can't have a persistent alarm going off in my house. I will get kicked out along with the printer. 

2. I have never had a prior problem with thermal runaway stopping the printer gracefully. Previously recoverable variances now cause a 100% failure. 

Reverting is the only sensible choice for me.

vvegastl, 2023-01-18:
It's a VERY long story, but I had a hotend completely burn up on a completely stock Prusa MK3S when thermal runaway failed after installing a previous 3.12 beta (instructed to do so by Prusa support to address another issue I've been having since buying the printer), temperature calibrated as support instructed. Prusa replaced the destroyed parts, sent new board to try, etc and suggested installing a follow-up 3.12 RC update, but that introduced the same thermal anomaly issues described here, and I reverted back to 3.11 and have had no issues since (beyond the one I contacted support about initially). 

Was browsing the forums after the 3.12 final update hoping that issue would be resolved (as 3.12 did appear to fix my original issue), but definitely not the case as the various TA accounts I'm reading has been exactly my experience. The protections in 3.12 are sound on paper, but it's clear that release needs more time for the bugs to be worked out and sticking/reverting with 3.11 is a smart (and safer) move until then.

mpkottawa, 2023-01-20:
I have the same issue after updating.  Stock Mk3s+ with e3d revo.  Tried multiple times, won't heat up nozzle

matthiazzz, 2023-01-27:
Are you guys too lazy to read release notes?

`In case you are running 3rd-party parts like the **Revo** hotend, the thermal model will report errors because it needs to be trained on a large set of samples. We welcome anyone who’s willing to help us with testing. Please provide log files in https://github.com/prusa3d/Prusa-Firmware/issues/3636.`

Rasmus-Fink, 2023-01-27:
Well, I consider myself pretty lazy but my issues is with a 100% stock, factory assembled unit...
//Fink

gold1640, 2023-01-27:
3.12.1.5686 worked

Installed 3.12.1.3686 and got a 'Thermal Anomaly' at the same point as 3.12.0, although the fan speed was different.

Was just thinking of pressing the reset button to try again when the thermal calibration continued and eventually came up as successful.

--
Ian Harris

closed 2023-01-27 by gold1640

thetwelvechairs, 2023-02-20:
Sorry, why is this in "Closed" state. Clearly there is an issue.

Rasmus-Fink, 2023-02-20:
Well, In my case there indeed was an issue with a defect temp sensor. After this was fixed the TM stopped complaining.
I think you might want to open a feature request instead that can more easily disable the TM or whatever you find nice...
My 10 cents...
//Fink

Prusa-Support, 2023-02-23:
This issue was closed by the author because s/he considered his/her issue solved.

.

In case of apparently inexplicable "thermal anomalies", you may want to check all the conditions that apply for the correct functionality of the feature in the complete introduction posted here https://github.com/prusa3d/Prusa-Firmware/pull/3552#issue-1324629390.

If you are running stock of Prusa parts, please get in contact with our official Tech Support for help.
https://help.prusa3d.com/article/customer-support_2287

If you are not using genuine Prusa parts, you may get hardly predictable results.
You may find existing issues relevant to your specific case under the label https://github.com/prusa3d/Prusa-Firmware/labels/thermal%20model.
If not, please consider opening a new specific issue that precisely describes your problem.

Michele Moramarco
Prusa Research

wifixcort, 2023-06-21:
Almost same problem with 3.12.2 firmware on a stock MK3 printer. It was printing perfect for a long time, today I upgrade the firmware and the message "Thermal anomaly" brings up. It can do the calibration steps ok, but when it start to print the message appear one time. There really is something going on with this firmware version!

The printer was never moved from its place, so the conditions are the same and the only thing that was changed was the firmware.

fink-at-trmc-dk, 2023-06-21:
I only saw the problem because I have an octopi connected. It shows a plot
of the temperatures and this revealed a bad sensor. The firmware was OK.
The sensor was not. (same HW as you btw)
BR
Fink

Prusa-Support, 2023-08-31:
Please don't take Thermal Anomalies lightly but consider repeating Thermal Model Calibration and carefully checking your print-head parts.
In the case of genuine Prusa parts, please reach out to our Customer Support for an evaluation of the problem.

Michele Moramarco
Prusa Research