prusa3d/Prusa-Firmware #3925: [FEATURE REQUEST] MK3S - allow disabling of "Temp model not calibrated yet" warning on startup

nagyben, 2023-01-16:
I'm running the MK3S with the Revo Six hotend which isn't compatible with the new temperature model (see #3890).

One new annoying side-effect of the temperature model is the need to manually acknowledge the warning `Temp model not calibrated yet` which comes up every time the printer reboots by pushing the rotary.

This is rather annoying when using OctoPrint. For example, when OctopPnt needs to reboot (e.g. after updating plugins), it due to the reconnect. With prior firmware this was fine, I could update Octoprint and continue my workflow without leaving my PC. Now I have to walk over to the printer and manually acknowledge the warning, then walk back to my PC to kick of the print using OctoPrint. 

I know this feels like a first world problem, but when your printer isn't in the same building (like mine) this becomes very tedious very quickly.

I have disabled the feature using the gcode as per the instructions on the [instructions](https://help.prusa3d.com/article/thermal-model-calibration_382488)

```
M310 S0
M500
```

But this does not remove the warning on printer startup which has to be manually acknowledged.

** Feature request **
Include some way of disabling the `Temp model not calibrated yet` message on startup so we don't have to manually ackowledge it every time the printer boots up

Thank you!

3d-gussner, 2023-01-18:
@nagyben This should been solved in FW 3.12.1-RC1 please verify and report back.

ZacharyACoon, 2023-01-28:
I was given a workaround to calibrate with a 75W heater that I think might apply to you.
https://github.com/prusa3d/Prusa-Firmware/issues/3963#issuecomment-1407484995

nagyben, 2023-01-29:
@3d-gussner I can confirm this is solved in 3.12.1-RC1

Thanks!

knightsljx, 2023-02-14:
This 'feature' should be removed until it collects enough data (for wide range of heaters on the market), is properly calibrated and implemented, and with an option to disable it.

arekm, 2023-02-14:
Bad thinking. Release early, release often is a much better approach. Without this released there wouldn't be any new data for any heaters other than prusa own hardware.

And a option to disable it *exists* already.  It's mentioned in first comment but/and works with 3.12.1+ firmware.

@nagyben please close this issue.

closed 2023-02-15 by nagyben