prusa3d/Prusa-Firmware #3931: [BUG] MK3S+ with Protoprint Raptor fails to calibrate thermal model

Rende009, 2023-01-18:
**Printer type** - MK3S + protprint raptor hotend - https://protoprint3dp.com/raptor
**Printer firmware version** 3.12

**Describe the bug**
  Attempting to start the new thermal model calibration WITH Raptor hotend immediatle there is THERMAL ANOMALY error with a warning tone but the calibration is still running to 230 on the second heating from 0 to 230 there is another thermal anomaly but this time the calibration is ended with error preheat the nozzle !.
  
**To Reproduce**
  Thermal model calibration with Raptor hotend with sock
  
**Expected behavior**
  Is it posible to disable thermal anomaly for calibration ?
  
[log.txt](https://github.com/prusa3d/Prusa-Firmware/files/10448165/log.txt)

wavexx, 2023-01-18:
Yes! Reading from the specs, looks like this is a 60W heater.
Run in sequence (one at a time):

```
M310 P60
M310 A F0
M310 S1
M500
```

This sets the expected power to 60W, runs calibration, enables and then saves the model data.
I'm curious about the results.

wavexx, 2023-01-18:
As an additional recommendation, if you didn't do this, is to run a PID calibration _first_ since you'll benefit for more accurate temperature regulation anyway. Disable the thermal model via ``M310 S0`` just before running it, then proceed as above.

leptun, 2023-01-19:
@Rende009 See https://reprap.org/wiki/G-code#M310:_Temperature_model_settings