prusa3d/Prusa-Firmware #3933: [BUG]<Hotend Fan spins full power>

Gamma4D, 2023-01-20:
**MK3S+** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**3.12.1-RC1-5682** -  [e.g. 3.8.1, 3.8.1-RC1, ...]

**no** - [e.g. MMU2S, MMU2, MMU1]
**no** - [e.g. 1.0.6, 1.0.6-RC2, ...]

**SD card**
  Please let us know if you print via SD card or USB/Octoprint

**Describe the bug**
After flashing the firmware, the hotend fan spins full power.
In normal case, except during booting, it becomes 4000RPM over 50C and 0RPM below 50C.
But now it spins about 8280RPM.
It produces a loud noise in printing.

**To Reproduce**
Even after rebooting and performing calibration again, it does not go away.
I don't know how to reproduce this error.

**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

3d-gussner, 2023-01-20:
@Gamma4D Thanks for reporting this issue.
Do you have a  stock printer or did you modify something?
If stock, can you see if your hotend fan is from Noctua or a Delta fan (more powerful and louder)?
Please also check in LCD `Settings -> HW setup` long press knob in the `Experimental` menu if `ALTFAN det.` is On or Off

H0Z0, 2023-01-21:
I also experienced this, with 3.12 fw the altfan det. might not work. It is turned ON in the experimental menu, but if I try to turn it OFF, after restarting the printer, it will be ON again. But it will not control the speed. It spins at 8280RPM.
However, it works fine on the same printer with 3.11 fw. The ON/OFF state remains there after the printer is switched on/off and also controls the speed.

gudnimg, 2023-01-21:
> It is turned ON in the experimental menu, but if I try to turn it OFF, after restarting the printer, it will be ON again.

I can confirm this on latest MK3 branch (3.13). MK3S+, the setting was turned ON, I toggled it OFF and then rebooted the printer via the CPU reset button on the LCD panel. And the setting was back to ON again.

Gamma4D, 2023-01-22:
> @Gamma4D Thanks for reporting this issue. Do you have a stock printer or did you modify something? If stock, can you see if your hotend fan is from Noctua or a Delta fan (more powerful and louder)? [...]

I have delta fan and ALTFAN det is on.

DominikPalo, 2023-02-05:
Unfortunately, I'm having the same issue - on FW 3.12, my Delta fan runs at the full speed... after downgrading FW back to 3.11, the fan is silent again. Unfortunately, I didn't realize that speed/noise increased at the time of the update, so after a couple of weeks of hearing high noise I decided to order a Noctua fan to replace it... but it wasn't needed.

goyetus, 2023-04-20:
Have three Mk3s+  , all with Delta Fan.
All in 3.11   ,  I cant go to 3.12  cause fan goes to 100%  and 8000 rpm ........

3.11 version is from 7 months ago......    What is the problem with delta fan?   Why is always enabled the ALT-FAN in 3.12 and above?   

Please,  could you give us a leadtime for fix this isssue?    I have delta fan couse  "noctua"  is not enough in my ambient hot country.......  

Regards  and thanks all the team for the hard work!!!

Prusa-Support, 2023-05-16:
I'm sorry, we don't have an estimated time for the fix. 

Not sure if this is relevant at this point but I'd like to reiterate that Altfan detection must be ON + a factory reset would be necessary to calibrate the PWM settings on the new fan. Otherwise, the fan will be working with possibly obsolete/wrong PWM settings.

>When a printer is run for the first time after the assembly, or [a full factory reset](http://help.prusa3d.com/article/factory-reset-mk2s-mk2-5s-mk3s_2216), the firmware will automatically detect the fan and set the appropriate PWM value. This value is stored permanently in the EEPROM memory and remains even after a firmware update. It can be only deleted by performing a full factory reset.
[...]
By turning the Altfan detection off, the printer will run the fan at 100% power.

https://help.prusa3d.com/article/experimental-menu-mk3-s-_161213#altfan-detection

Could you please confirm the status of _altfan detection_ as it is now?
Also, try to intentionally modify it and reboot the printer to see if the modified status remains or will be reset after the startup.
You may want to repeat the tests with 3.12.2 and 3.11.0 firmware versions.

Michele Moramarco
Prusa Research

Gamma4D, 2023-05-27:
> 

We can't turn off Altfan det. After turning off Altfan det and rebooting, Altfan det becomes 'on'.

Prusa-Support, 2023-06-20:
Minor voltage fluctuations have been seldom observed on some mixed combinations of hardware (EINSY boards, fans, thermistors... maybe PINDA probes and other parts too), due to minor manufacturing differences (yet within the specifics) but also and especially depending on the state of the hardware ("aging").
These minor fluctuations in rare cases may cause signal noise within the circuit, but this is usually totally negligible and will sporadically manifest as temporary glitches.

We may be looking at a very specific edge case of voltage fluctuations.
Users who have multiple printers or spare parts may try to swap parts (else than the hotend fan) and report if there is a specific piece of hardware that causes the problem (or a specific combination of hardware parts).

I'm sorry for the annoyance. This seems to be a complex hardly reproducible condition.
We keep investigating.

Michele Moramarco
Prusa Research

goyetus, 2023-06-20:
Thanks for the update @Prusa-Support  (Michele Moramarco)  

In my case ,  probe started failing after  the original Delta Fan crashed.    
I tryed like 5-6 different Delta Fans ,  but SuperPinda keeps failing  while Fan is running. 

I send you by email about 1 month ago,  several videos with the problem. 
I can´t swap more  hardware between my 3 prusa mk3s+ ,  I need them online and working :)  

Thanks a lot for the update and the hard work.

goyetus, 2024-02-07:
Ok,  you close an Unresolved issue  affecting all the Fans I purchased from Prusa.
Is going prusa to refund me all the spare parts  I purchased in Fans?   About 6 of them.   

I think I have been patient enough.

Prusa-Support, 2024-07-09:
Could you please try again with the new firmware release [3.14.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.0)?
This problem is hardly reproducible so your feedback would be most appreciated.

Michele Moramarco
Prusa Research