prusa3d/Prusa-Firmware #3951: [BUG] Nozzle change doesn't unload the filament

DominikPalo, 2023-01-25:
**Printer type** MK3S+
**Printer firmware version** -  3.12.0

**MMU upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]

**Describe the bug**
According to the [recommended procedure](Nozzle change), the filament has to be unloaded AFTER preheating to 280 °C. Unfortunately, when I use the new "Nozzle change" functionality, the filament is not automatically unloaded after the final temperature is reached (and it is not even possible to unload the filament manually - as the menu is locked during the "Nozzle change" procedure). A workaround is to restart the printer, unload the filament manually and start the "Nozzle change" procedure again...
  
**To Reproduce**
  1) Execute "Settings" -> "HW setup" -> "Nozzle change"
  2) Wait until the hotend reaches 280 °C
  3) Filament is not unloaded
  
**Expected behavior**
  1) Execute "Settings" -> "HW setup" -> "Nozzle change"
  2) Wait until the hotend reaches 280 °C
  3) If filament is present, is it automatically unloaded

mix579, 2023-01-25:
Yep. You have to unload the filament before you initiate the nozzle swap. The (old) instructions are incorrect. Plus, it never made sense to unload PLA or PETG at 280…

Prusa-Support, 2023-02-03:
Hello! Thank you for reporting this, the article is now updated 🙂 

Alessandro Pantaleo
Prusa Research

closed 2023-08-11 by Prusa-Support