prusa3d/Prusa-Firmware #3952: [ENHANCEMENT] Show temperature with new nozzle change procedure

mix579, 2023-01-25:
I like the new menu option to initiate a nozzle swap. However, once heated up to 280 degrees, it doesn't show me the temperature anymore. It is common for the temperature to drop while removing the old nozzle and inserting the new one (from holding a spanner to the heater block, or a torque wrench etc). I then wait for the temperature to back to 280 before tightening the nozzle with a torque wrench. 

With the current procedure I have no idea what the actual nozzle temperature is, so I just wait for some time assuming it'll be back at 280. It would be very helpful to show on the screen the actual temperature.

3d-gussner, 2023-01-26:
@mix579 Thanks for the report, we will look into this.

mix579, 2023-08-10:
Still as valid as it has ever been