prusa3d/Prusa-Firmware #3954: [BUG] MMU firmware hex file is missing from the MMU zip

adamdport, 2023-01-26:
https://help.prusa3d.com/article/firmware-updating-mmu2s_155118 says

> The filename of the firmware for the MMU2S is 'prusa3d_fw_MMU2board_x_x_x_', where the x is the firmware version number. Do not select the MK3S firmware located in the same folder.

However, the latest MMU zip file 3.12.1 does not contain such a file. https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_12_1_MK3S_1_0_6_MMU2S.zip

Is the documentation wrong or is the zip file wrong? Thanks.

3d-gussner, 2023-01-26:
@adamdport Thanks for the report. I checked the zip and indeed it is missing, forwarded the issue.
 
The MMU2S firmware is still 1.0.6 so if you have one no need to update it.
In case you don't have it  
[prusa3d_fw_MMU2board_1_0_6.zip](https://github.com/prusa3d/Prusa-Firmware/files/10512749/prusa3d_fw_MMU2board_1_0_6.zip)

JakoobCZ, 2023-01-26:
Hi Adam, this was a mistake. Zip archive repacked with this missing MMU firmware file and currently being uploaded to a server. Give it please some time before it gets live. Ideally also delete cache for help.prusa3d.com before you attempt a new download.

adamdport, 2023-01-26:
Oh, I didn't realize that the MMU wasn't updated. The MMU-specific changelog shows many many updates that I guess are mostly just duplicates of the MK3S changelog for some reason? 

https://help.prusa3d.com/downloads/mmu2/firmware

Not sure what's going on there. thanks.

Prusa-Support, 2023-01-31:
Hello @adamdport, I confirm it - the changelog on the linked page is related to 3.12.1 for MK3S+ compared to the previous version - the reason is that on that page, we publish both MK3S/+ and MMU2S firmware, so we'd report changes for both of them. 

Alessandro Pantaleo
Prusa Research

gudnimg, 2023-07-04:
Since there is no direct firmware/code change required to resolve this issue, can this Github issue be closed?

adamdport, 2023-07-04:
sure!

closed 2023-07-04 by adamdport