prusa3d/Prusa-Firmware #3959: [BUG]  M112 called upon print finish, when print head lifts up

kristapsbasovs, 2023-01-27:
**Printer type** - MK3S+
**Printer firmware version** - 3.12 and 3.12.1

**SD card or USB/Octoprint**
Octoprint

**Describe the bug**
 M112  called when the print finished and the print head just lifts up. I suppose the thermal model is too sensitive or something. Happened on both 3.12 and 3.12.1 with two different printers. Both of them are running bear extruders and silicone socks. This is definitely not a hardware issue as one of these printers received a new original heater and thermistor today. All printers are properly calibrated, including PID and thermal model. Before M112 there are no thermal anomaly beeps, on 3.11 these printers are working great. There are no visible temperature drifts on LCD or Octoprint.  It does not happen all the time, but frequently enough.

3d-gussner, 2023-01-27:
@winmeabrick Thanks for reporting. Can you please enable serial logging add the OctoPrint serial.log? See the OctoPrint part https://help.prusa3d.com/article/crash-dump_364959

sarusani, 2023-01-28:
Are you  sure it only happens on 3.12.x? I had a similar problem caused by a bad USB cable. It disconnected OctoPrint on certain printer moves because the printer cable brushed against the USB cable. After multiple connection losses OctoPrint sends a M112.

If you check the log as 3d-gussner suggested, you should be able to see if the M112 was sent from OctoPrint.

winmeabrick, 2023-01-28:
@3d-gussner I've now switched back to the 3.11 as I use printers 24/7 for commercial purposes. I'll try to flash back 3.12 next week and replicate the crash sometime in the next few days.

@sarusani Yes, pretty sure, as they were stable on 3.11 before 3.12.x and now I went back to 3.11, and again - works great. Also, I run Pi Zero 2W with GPIO, so no USB cable.

wseifert, 2023-01-30:
For me the "new" feature in 3.12.x was build with hot knitting needles, not ready for the public. I have also this error (and some others originated by the new feature). For me this feature is total unnecessary, I have my MK3S since 2 1/2 years and had no thermal issues. I am switchinf back to 3.11

Prusa-Support, 2023-02-28:
@winmeabrick. Our developers would gladly have a look at the logs. No pressure though.
I'd like you to take this chance to possibly review the complete introduction to the feature and its expected working scenarios here https://github.com/prusa3d/Prusa-Firmware/pull/3552#issue-1324629390.
I'd also like to highlight that the results are influenced by the different extruder designs (print fan especially) and the ambient conditions (large range of ambient temperature).
Please repeat the calibration if you put or remove the silicon socket, and be mindful that the thermal model was not specifically tested on Bear Extruders. In most cases, a re-calibration could help though.

Possibly follow (or repeat) these basic troubleshooting steps.
- Perform a new [PID tuning](https://help.prusa3d.com/article/pid-tuning_2265) in the first place.
- [optional but recommended] Consider moving the print-head close to the heatbed to introduce higher "thermal stress" during the thermal model calibration, rendering the thermal model a bit more "resilient".
- Repeat the [thermal model calibration](https://help.prusa3d.com/article/thermal-model-calibration_382488) after.

.

We respect the opinion of everyone, greatly appreciate comments, and find our community suggestions most valuable, but we would like to avoid misinformation.

The feature was developed over 150 000 samples and posted as Pull Request on 1 Aug 2022 (PR [3552](https://github.com/prusa3d/Prusa-Firmware/pull/3552#issue-1324629390)).
The first beta release came out on 29 Sep 2022 ([3.12.0-BETA1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.0-BETA1)).
The final release was posted on 11 Jan 2022, almost half a year after the PR ([3.12.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.0)).

The early development stages remain behind the scene but, for peace of mind, I hope this can be evidence of a thoroughly tested and developed feature.
This feature is for advanced printing safety. It will spot barely acceptable and even slightly damaged/misaligned heaters/thermistors, and possibly define whole new quality standards.

Firmware downgrading or switching the feature off remain viable options (that we don't recommend by any means).

Michele Moramamarco
Prusa Research

Prusa-Support, 2023-03-02:
Not sure if related, but the recently released [FW 3.12.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.12.2) version improves the Thermal Model compatibility over serial connection and with some third-party hardware.

Please give it a try.

Michele Moramarco
Prusa Research

closed 2023-11-14 by github-actions[bot]