prusa3d/Prusa-Firmware #3961: [FEATURE REQUEST]  settings backup and restore to sd card

cnlson, 2023-01-27:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]

**Is your feature request related to a problem? Please describe.**
  A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
  
**Describe the solution you'd like**
  A clear and concise description of what you want to happen.

3d-gussner, 2023-01-28:
Thanks for the idea but as @wavexx already explained in https://github.com/prusa3d/Prusa-Firmware/pull/3957 we are always short on resources.

I tried long time ago to contact an OctoPrint Plug-in author (as I am not familiar developing around OctoPrint and Plugins) if he would be interested to implement ab backup/restore/modify Plugin.

If there is someone willing to develop an OctoPrint Plug-in, please contact me if you need more information about it.

As we store all/most information in the EEPROM and provide a "D-code" `D3` to read/write values it should be possible to implement this.
The eeprom values are documented [here](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/eeprom.h#L42-L369) and [here](https://3d-gussner.github.io/group__eeprom__table.html) 

Modifying the EEPROM manually is of course **AT OWN RISK** and the usage of `D3` write command incorrectly may result in state where ONLY a factory reset helps. I know what I am talking about, happened several times to me (typo or to fast enter)

cnlson, 2023-01-28:
It should be in prusa slicer everyone has that not everyone has octoprint.

OK to close this and I'll take my suggestion to the prusa slicer forum

3d-gussner, 2023-01-28:
Good luck with PrusaSlicer :crossed_fingers: 
I am not sure but PrusaLink may get at some point such a feature.

alexiri, 2023-01-29:
The logical place for this to happen is as part of Octoprint's [Firmware Updater](https://plugins.octoprint.org/plugins/firmwareupdater/) plugin. Ideally, I was thinking the firmware updater would automatically take a backup of the EEPROM before an upgrade and allow you to restore from previous backups if you needed to.

I was actually going to try to tackle this, but I ran into a pretty big problem: `D3` is not a standard gcode command, only Prusa firmware implements D-codes. In fact, the [reprap G-code wiki](https://reprap.org/wiki/G-code#D:_Debug_codes) has this to say:

> NOTE: These D codes are not compatible with the Gcode spec ( NIST Gcode spec, initially written for CNC, which Reprap initially messed up, after what the CNC side of Reprap has spent years trying to get Reprap back to respecting it ), and will make life hell for those building machines that do more than just 3D printing, or controllers that support this, as well as possibly cause other issues. I see this is from Prusa, and I know these guys are really cool about Open-Source, so I'm sure they care a lot about respecting standards and other projects as well, so I'm certain they'll fix this as soon as they see there is an issue. Thankfully this is very easy to fix, just change from D1 to Mxxx D1, where you find a Mxxx Gcode that is currently free and start using it. In Gcode, only M, G and T are keywords, D is a parameter, you can't make it a keyword without messing everything up. I'm sure for your subset of using Gcode, that's probably just fine, but it's not for the rest of us, and the whole point of adding things to this page, is for others to use it if they want/can/need, and the way it currently is ( breaking a standard ), this can't be used by others.

How come Prusa is using these non-standard commands? Could they be replaced for `Mxxx Dy` as suggested? I guess this functionality has a higher likelihood of being adopted by others if it respects the standard rather than being completely incompatible with other firmwares.

wavexx, 2023-01-29:
> I was actually going to try to tackle this, but I ran into a pretty big problem: D3 is not a standard gcode command

Is this a real problem though? We already have 3d-printer gcodes flavors which are not strictly compatible with each other. Although you can have a base which works for _most_ things (I think almost every FW _tries_ to), when you're going to look at low-level details this is only going to get more and more different.

To use D3, for example, you're relying on the fact that this performs a raw memory read from the eeprom, which just so happens to contain the settings. This allows to poke at the low-level detail of the machine, and it's not intended to work across different machines at all. The FW lets you poke/mess with the system, but attempts in no way to make this detail portable, or even safe to use.

alexiri, 2023-01-29:
I wouldn't expect a FW dump to be portable to any other machine, of course. I was just thinking that if I wanted to add this functionality to Firmware Updater, the maintainer is going to point out that it's a feature that won't work on any other FW version. If other firmware versions were to adopt `D3`, it'd be nice if it wasn't totally breaking the standard.

wavexx, 2023-01-29:
If I was the maintainer of "Firmware Updater" my main concern wouldn't be D3, but the fact that this is only useful for the MK3 FW and in a very limited set of cases. Not only you can't port this over to another FW, but you need to ensure that when restoring the settings you're also uploading it to the same FW version that made it so this might require extra burden on the plugin for a feature which is only used for testing. This definitely goes beyond the usual "firmware update" logic.

alexiri, 2023-01-29:
Fair enough, you've convinced me not to implement it. :rofl: 

In any case, do you know the story behind the D-codes and why Prusa breaks the "standard" with them?

wavexx, 2023-01-29:
No idea :), but I did try to plan some new gcodes with Marlin 2.x and RepRap and in some of the discussions we had the conclusion was "just make up a new random number". There's very little argument consistency in 3d-printer gcodes as well: usage of common letters such as ABCD/F/etc is mostly random for a large number of M codes. Only 'G' is somewhat consistent.

I think 'D' makes perfect sense for something you intentionally declare as non-standard/non-portable. It doesn't conflict with any potential M code of other FWs we can copy later.

cnlson, 2023-01-30:
@alex iribarren prusa is leading the way, not really breaking the standard.
eventually marlin will have this feature. Just as with other features the
mk3 had first and then marlin absorbed. iirc linear acceleration maybe?

arekm, 2023-01-30:
Putting that in Octoprint's Firmware Updater is a bad idea. Firmware Updater is barely maintained (and doesn't even support proper firmware upgrade for Prusa (no support for languages)). Putting this there wouldn't give us any benefits and only maintenance problems.

Separate plugin is a best place. See https://plugins.octoprint.org/plugins/eeprom_marlin/ and https://plugins.octoprint.org/plugins/eeprom_repetier/ as examples.

Also in OctoPrint plugins can provide API for other plugins. So for example, later, Firmware Updater could be made to detect if OctoPrint-EEPROM-Prusa (example name) is installed and use its capabilities on upgrades etc. 

Octoprint has hooks that allow firmware variant to be recognized detected, so it can be made to detect if we are working on prusa printer and it can show warning if firmware version doesn't match version that was there when settings were saved etc.

btw. PrusaSlicer also isn't best... for example I have printer that's available only remotely, without PrusaSlicer connected to it. So some don't have OctoPrint, some don't have PrusaSlicer connected.

Prusa-Support, 2023-03-01:
I may seem to see a correlation between this issue and issue #2978.

Maybe one or the other could be closed, but, for now, I find it worth a mention for linking them. It may be useful for future GitHub explorers interested in firmware values backing up.

Michele Moramarco
Prusa Resaerch

closed 2023-11-14 by github-actions[bot]