prusa3d/Prusa-Firmware #3963: [BUG] [CRITICAL] [3.12.1] 75W heater.  Wizard calibration appears to ignore temperature anomaly disable with M310 S0\nM500

ZacharyACoon, 2023-01-28:
75W heater.  (yup)

1. Followed the instructions to send `M310 S0` and `M500` over serial to disable thermal anomaly detection.
https://help.prusa3d.com/article/thermal-model-calibration_382488
2. Initiate calibration wizard.
3. Fails with "THERMAL ANOMALY"

Cannot proceed.

wavexx, 2023-01-28:
This is expected. `M310 A F1` will check against the stock V6 heater (and complain that the heater is too powerful).
Is this a V6 with a 75W heater? In that case you have to set the power:

```
M310 S0 ; disable
M310 P75 ; set heater power
M310 A F0 ; calibrate for a different hotend
M310 S1 ; enable again
M500 ; optionally save if everything is working
```

ZacharyACoon, 2023-01-28:
> This is expected. `M310 A F1` will check against the stock V6 heater (and complain that the heater is too powerful). Is this a V6 with a 75W heater? In that case you have to set the power:
>
> ```
> M310 [...]

This worked.  I'm going to spam this on some related issues to unblock people.

ackerthehacker2, 2023-02-02:
If you disable thermal protection, does running calibration re-enables it?  Also, you might want to hilight in the documentation running PID calibration first as I think it will help with a lot of problems folks are having with non-moded printers).

3d-gussner, 2023-02-02:
@ZacharyACoon Please consider to close this issue.

jdrozdz, 2023-02-22:
> This is expected. `M310 A F1` will check against the stock V6 heater (and complain that the heater is too powerful). Is this a V6 with a 75W heater? In that case you have to set the power:
>
> ```
> M310 [...]

Not working for me

closed 2023-04-04 by Prusa-Support